meshtastic/firmware #3989: [Bug]: PicoW GPS Firmware

OskarZyg, 2024-05-29:
### Category

Hardware Compatibility

### Hardware

Raspberry Pi Pico (W)

### Firmware Version

2.3.7.30fbcab

### Description

The Pico W works just fine with GPS, but the firmware doesn't allow it to support it.
It is already supported on [picow-slowclock](https://github.com/meshtastic/firmware/pull/3412). 

The changes I made to get it to work were:
GPS.cpp: set condition to true (in reality this would probably be defined(PICO) or whatever)
variants/rpipicow/variant.h: added #define HAS_GPS 1

I am willing to write the PR myself later on

OskarZyg, 2024-06-02:
I am experiencing issues with checksum failures. If anyone else can reproduce that would be helpful.

A more detailed explanation of the symptoms from the discord (thanks cynfab_63112):
> you are losing data somehow, things like this--> $GPGGA$GPRMC,,V,,,,,,,,,,N*53 <-- indicate that the last part of the GGA sentence was overwritten by the following RMC sentence. This is a recurring problem in your whole post. IMHO TinyGPS++ isn't smart enough to disgard the first partial sentence and pick up on the next. Checksum failures result. Now the big question is what can be done about it.The root cause maybe too small a serial RX buffer, or some other software related issue. Probably best to invistagate what other RP 2040 varients are doing, do they work without checksum failures or not.

Mictronics, 2024-08-20:
I have seen checksum failures when implementing Waveshare L76B GPS support on the Pico. Solution is an increase of serial FIFO size.
See my pico_waveshare branch in my fork: https://github.com/Mictronics/meshtastic_firmware/commit/766287d3fddde8c49065534e069ad1261422e19b

What GPS you are talking about?

OskarZyg, 2024-08-22:
I believe I was testing with a u-blox 7. I'll retry with the increased FIFO buffer size, I wasn't aware that I could do that (i don't have much experience with embedded programming).

fifieldt, 2024-09-16:
@Mictronics ,did you have time to send in that FIFO size patch, or want help?

Mictronics, 2024-09-16:
@fifieldt The support for this patch is not fully implemented in official firmware. The code transfer from my fork into meshtastic official was done by caveman99. Change of the Serial interface class was either not included or modified again after.
Till now I hesitate to create a PR since I don't know what impact it has on the remaining GPS code.
Also, the official firmware doesn't make a difference between Pico and Pico Waveshare, like I do in my fork+branch.

closed 2024-09-17 by thebentern