meshtastic/firmware #4007: [Bug]: Factory reset not actually resetting node.

madeofstown, 2024-05-31:
### Category

Other

### Hardware

Heltec Wireless Tracker

### Firmware Version

2.3.10.d19670b

### Description

Performing a factory reset from the Android app on Heltec Wireless Tracker v1.1 appears to leave old messages and nodes (now all unknown name) on the device. 

Steps to reproduce:
1. Be connected to node via Bluetooth to your Android device. 
2. Go to settings and select the Factory Reset option. Confirm. Whatch node reboot.
3. Re-pair and reconnect to same Android device.
4. See message on node screen. It's from ??? 2 hours ago.(I did receive the message 2 hours ago before driving to a different city on different side of mountain range and factory resetting. Pretty sure it's not being retransmitted.)
5. Scroll through screen and see a bunch of Unknown Name nodes.

clwgh, 2024-06-01:
It may be that you're seeing those nodes and messages stored on the Android app (now unknown since the node has indeed factory reset). There may well be a separate option in the Android app to clear the node database from the app.

To be sure, you can check the nodes on the device directly using the Python CLI, with the command `meshtastic --nodes`. If you see none (or a handful in the seconds since the factory reset) then it is resetting as expected. If you see them all still there then it's not.

madeofstown, 2024-06-02:
Clearing the node database before or after factory reset doesn't seem to make any difference.

djripcord, 2024-06-16:
> It may be that you're seeing those nodes and messages stored on the Android app (now unknown since the node has indeed factory reset). 

I don't believe the android app perpetually stores nodes. They seem to be stored in a volatile fashion. They will stick around when disconnected from a device, but when reconnecting, it pulls down a fresh copy of the node DB stored on the device. Regardless, it's not possible to narrow down the issue to app or node unless @madeofstown provides the output from the CLI on a node query.

madeofstown, 2024-06-17:
Sorry, I've been working on other projects. I'll try to get around to capturing CLI output from the node this week.

fifieldt, 2024-09-16:
Hi @madeofstown , we've made some changes to factory reset over the past couple months. Could you give it a try again with the latest firmware?

PR #4982 (2024-10-07): Remove has_rx * on installDefaultDeviceState
Closes #4007
closed 2024-10-07 by fifieldt