meshtastic/firmware #4012: [Bug]: CDEBYTE EoRa-S3 Not Booting without USB Host Serial

ndoo, 2024-06-01:
### Category

Hardware Compatibility, Serial

### Hardware

Other

### Firmware Version

2.3.10.d19607b

### Description

Connect CDEBYTE EoRa-S3 to 5V charger (not host computer) using USB-C to USB-A or USB-C to USB-C cable.

Device does not boot and OLED never displays any output. User LED does not light up or blink.

As this board uses the ESP32-S3's native USB pins (schematic is in the [EoRa-S3-900TB downloads page](https://www.cdebyte.com/products/EoRa-S3-900TB/4#Downloads) -> hardware data), I suspect the issue and fix would be related to a whole bunch of USB connectivity issues the T-Beam Supreme also had, due to that board also wiring the USB-C port to the native USB pins of its ESP32-S3: #3944 #3912 #3856 #3736 #3597.

nikvostro, 2024-06-02:
I've got the same behaviour for Lilygo T3S3 v1.2 for 3 devices I've flashed with 2.3.10 Beta just recently. Devices doesn't respond anyhow (power on/off, HW reboot etc) if NOT connected to PC.

blurcak, 2024-06-06:
I have the same problem on my  lilygo t3s3 v1.3  via webflash.
Device boots only when connected to PC. Any configuration change is ending in "Rebooting" and device starts only after disconn/conecting the usb.
FW 2.3.4 and 2.3.6 are working OK.

ndoo, 2024-06-06:
Updated the issue title to be more generic to ESP32-S3 with native USB

thebentern, 2024-06-07:
Please test these artifacts and see if it resolves the issue:
T3S3 - https://github.com/meshtastic/firmware/actions/runs/9418801324/artifacts/1579655956
EoRA S3 - https://github.com/meshtastic/firmware/actions/runs/9418801324/artifacts/1579647484

blurcak, 2024-06-07:
Hi, i was not able to install it on my T3S3, i tried only webflasher, but it was stuck on "stub running ... "
I have new PC so no python etc. instaled yet.
2.3.11 alfa works ok

thebentern, 2024-06-07:
>2.3.11 alfa works ok

Wait, so it's already working in the latest alpha release of the firmware?

thebentern, 2024-06-08:
This is working fine in the latest 2.3.11 alpha. I just tested from full erase and install on a T3S3 and EoRA S3

closed 2024-06-08 by thebentern