meshtastic/firmware #4013: Nano G2 Ultra on doesn't shows direct message

miodzicho, 2024-06-01:
### Category

Other

### Hardware

Other

### Firmware Version

2.3.6.10

### Description

When sent direct message to Nano G2 Ultra, message doesn't appears on screen. On any screen. Only received stations are visible.

neilhao, 2024-06-02:
Previous versions of the Meshtastic firmware may save stale data, causing devices to get stuck in a crash loop during startup. If you experience issues when upgrading your nRF52 device from a previous version of Meshtastic, you may need to perform a full factory reset of the internal flash memory. 
You may want to [factory erase your nRF52 device](https://meshtastic.org/docs/getting-started/flashing-firmware/nrf52/nrf52-erase)before continuing to [flash firmware](https://meshtastic.org/docs/getting-started/flashing-firmware/nrf52/drag-n-drop#flash-firmware)

caveman99, 2024-06-03:
did you enable the Store&Forward Module? It does not have any effect on the Nano G2 Ultra or any other nRF52 based device, but switches off the display of received messages.

closed 2024-06-06 by thebentern