meshtastic/firmware #4016: [Feature Request]:  Root topic by channel or similar solution …

memo-567, 2024-06-02:
### Platform

other

### Description

“Root Topic” and channels in Europe, here using Germany as an example

It certainly applies to other countries as well, but from my own experience I can only speak for Germany.

Before switching to firmware 2.3.2 there was only the root topic “msh”. All channels, the regional channels and also the transregional channel "Deutsch", were based on this. Communication with each other went smoothly.

After the switch to 2.3.2 and higher, the addition of the frequency was introduced. What went smoothly after an update for countries with only one frequency in use - it just had to be updated - led, at least in my impression, to a mess in Germany (Europe?).

There are now not only users who still use older firmware versions and users with the new firmware, but also different root topics.

So here, I'll take the channel "Deutsch" as an example, there is not only still msh/2/c/Deutsch, but also the variations msh/2/e/Deutsch, msh/2/e/EU_866/Deutsch and msh/2/e/EU_433/Deutsch.

It would be nice if we could find a solution here so that communication between all nodes would be possible again in the sense of emergency and crisis communication and prevention.

garthvh, 2024-06-02:
They have to break down into smaller groups, that is what the topic is there to do.

The c is old firmware and was a mistake, it has a very short remaining shelf life.

memo-567, 2024-06-02:
Unfortunately, that's not how it works in reality. As soon as different root topic are set, cross-communicating with users of a different setting is no longer possible.

It would be nice if two regions could continue to communicate with each other, to warn of dangers such as flooding and the like, regardless of frequency.

Unfortunately, it doesn't look like that at the moment. If region A has decided to continue using "msh" as the root topic, but the neighboring region has decided to use msh/EU_868 or msh/EU_433, then communication is no longer possible. That's probably the situation in Germany at the moment.

memo-567, 2024-06-02:
The update from c to e is of course logical and inevitable.

memo-567, 2024-06-02:
If setting the root topicps via channel is not possible, then perhaps msh/EU or something similar would be a possible overarching solution.

garthvh, 2024-06-02:
It is just a string, so you can use whatever string you want, the goal of the presets is to get away from flooded global topics.

closed 2024-06-02 by garthvh