meshtastic/firmware #4030: [Bug]: missing integrity checks let attacker forge arbitrary message content using a known plaintext attack and replaying messages even when PSK is not known

Jorropo, 2024-06-03:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

all ¿ (foundemental protocol issue)

### Description

This was discussed with @caveman99 in `#contributor-lounge` he gave me the go ahead to post it here.

Here is pseudo code targeting the admin module, this would let an attacker observe an admin packet, modify it and change more or less any settings at will on all the nodes using this admin PSK:
```py
def encryptPacket(nodeid, packetid, key, message):
 bitstream = generateAesCtrBitstream(uint64bytes(packetid) + uint32Bytes(nodeid) + uint32bytes(0), key, len(message))
 return bitstream ^ message

decryptPacket = encryptPacket

setOwnerPacket = createSetOwnerPacket(to, newName) # this assume this has predictable output, the attack is harder if this does not, but not impossible, the main drawback is that it would require many round trip with modified packets to guess the order of fields
packetId = generatePacketId()
cypherText = encryptPacket(sourceNodeId, packetId, key, setOwner)
sendLora(sourceNodeId, packetId, cypherText)

# let's assume I am a an attacker, all of the following code run on my computer, I just observed some packet I couldn't decrypt it, and I then I see a new nodeinfo with a changed name
# all of this to say, let's assume I was able to learn `to` and `newName` from contextual clues
# I also know `sourceNodeId`, `packetId` and `cypherText` since theses are sent in clear text
setOwnerPacket = createSetOwnerPacket(to, newName) # recreate the plaintext, there are many other ways to achieve this even when the whole content cannot be guessed by doing trial and error
bitstream = cypherText ^ setOwnerPacket # the whole magic is happening here
# I can now forge abitrary content that is len(bitstream) long
maliciousPayload = createSetChannelPSK(to, index=1, psk=PSKControlledByAttacker) # assume this is setting admin channel psk
maliciousPayload += generateUnusedProtobufFieldsPadding(len(bitstream) - len(maliciousPayload)) # this abuse how protobuf is forward compatible, if I send a bytes buffer with an unused protobuf id the node will ignore it on purpose, this mean the node will skip any remaining bytes
cypherText = bitstream ^ maliciousPayload # encrypt it back
sendLora(sourceNodeId, packetId, cypherText) # replay spoofed packet with modified content
# you might need to send other packets first to make the node forget it has seen this packet else it might disregard it as duplicated
# you could also jam the original packet so it never reach the target node
```

---

**I recommend anyone using the remote administration module disable it until a fix is implemented.**
Note: this attack require capturing valid admin packets first, so if the module is turned on but you never used it you are not affected, I still think you should disable it because you might forget and use it in 6 months but who knows.

---

The fix I recommend is to use AEAD.
I have prior experience with AES-GCM it is AES-CTR (what we are currently using) with a bit of extra math, this adds a 16 bytes hash after the message, it is accelerated on ESP32 and the extra math is not extremely expensive.
The main drawbacks:
- The biggest admin packets don't have enough room for a 16 bytes auth.
  This is not a very hard engineering problem, protobuf isn't that efficient and I guess we could find a way to gain a few bytes.
- Backward compat / extra 16 bytes cost on all packets.
  This can be solved by making this optional when configuring your channel, most channels like `AQ==` would gain nothing but we could enforce this for `admin`. So users would be able to toggle if they want integrity protection 
- nrf52 chips don't support AES-GCM acceleration, they do support AES-CCM tho, which also do AEAD, I am not familiar as familiar with AES-CCM but my limited opinion is that it should be good enough, it is based on AES-CBC instead of AES-CTR which GCM is based on.
  ESP32 chips also support AES-CCM acceleration.
  So AES-CCM should be a good alternative.

AES-CCM is often what is used in WPA3.
AES-GCM is used in an overwhelming of TLS1.3 connections (altho Chacha20-Poly1305 is also often implemented as a fallback for CPUs who lack AES hardware acceleration), updated TLS1.2 clients also often prefer AES-GCM based suites. You are extremely likely to be using AES-GCM to view this very same issue.

This would not solve all the issue present in meshtastic's encryption, this only solve Integrity but not Perfect-Forward-Secrecy or Authentication.

### Relevant log output

```Shell
N/A
```

Jorropo, 2024-06-03:
If implementing AEAD is not possible, a correct MAC (which include both message content and PSK) is also acceptable, this historically created Side-Channel-Attacks in TLS1.2 which required TLS1.2 implementations to add mitigations and TLS1.3 removed all non AEAD support.

Even in the worst correct case MAC with SCA is way better than the current state of things.

Jorropo, 2024-06-04:
Actually I over-complicated padding.
Message length is not encrypted (it is stored in the explicit LoRa header), you can simply truncate the message to the length you want.

xnyhps, 2024-06-04:
> this adds a 16 bytes hash after the message

16 bytes is ideal, but shorter authentication tags can be used for AES-GCM/AES-CCM. 12 bytes would still be fine, and depending on the situation, even 8 bytes can be used. The slow rate at which messages can be sent could justify using shorter tags.

xloem, 2024-06-04:
If the fundamental protocol will change, given issues like this have been so extant, I would propose finding existing cryptographic solutions and systems to use as much as possible, so as to share work in maintaining them with others. There is AI-driven attack research now.

EternityForest, 2024-07-01:
#2610 is possibly relevant.

As I understand it, MACs are *not* subject to the birthday paradox (Reference: https://crypto.stackexchange.com/questions/50085/are-macs-vulnerable-to-birthday-attacks).

An 8-byte MAC, at 500 messages per second, would take hundreds of thousands of years to get one fake message through, if I understand the math: 2^63÷(500×3600×24×365)

A four-byte MAC at a still unrealistically high 10 messages per second would take 6 years. But, you could lock out for a few seconds if you get too many incorrect MACs, so it would be more like hundreds of years.

I think a more relevant attack would be replay attacks.  Not likely for the admin channel, but very likely in some use cases.

I've mentioned this before in other threads, but rather than an 8-byte MAC, what about a 4 or five byte MAC, with a timestamp?

Setting the time could be done with a dedicated time sync packet, itself authenticated.  If you don't get a time update in a few months, fall back to ignoring time.

The timestamp could actually just be implicit, you could make the MAC key a hash of (Timestamp+actual key).   Then you only need the four extra bytes.  

The timestamp would just have to only be precise to a few minutes, and you'd have to try decrypting with both possible keys, if you're right on the edge of a time boundary.

Or, you could have an explicit timestamp, in the plaintext associated data, which would incidentally also serve as a time sync packet, that any other node could also read, just without the authentication.

For replay attacks *within* few minute time window, a simple one byte message counter counter be used, that resets at the start of a time window.  Just keep track of the time you last got each of the possible values.

It breaks down if you have multiple senders to one receiver within a time window, but that's fine for admin channel,  and most home automation tasks, and could be disabled for the rest.

If nodes randomize their starting counter values it would mostly work with multiple nodes anyway.

With more memory you could also just keep track of all messages you got in the last few minutes.

Jorropo, 2024-07-06:
Ok so I was implementing CCM-optin and https://github.com/rweather/arduinolibs/issues/52 is a blocker for linux-native and rp2040 targets. Chances are I'll fix this upstream.

fifieldt, 2024-09-16:
Hi @Jorropo . In 2.5.0 we've re-done the admin channel with some PKI , which should fix this issue.

https://www.youtube.com/watch?v=7q-dNZzpUlM

xnyhps, 2024-09-16:
I'm a bit confused why this issue is reported only for the admin channel, and (as it appears currently) only addressed for the admin channel. Any private channel would be affected by this in the same way. Of course, the admin channel is the most tempting as it allows re-configuring a node, but even on other channels it would be an issue.

I could spin up a repeater that always flips a few bits before forwarding a message. Any receivers who got the message via this repeater won't be able to determine that the message was manipulated (unless it corrupts the protobuf structure, but I can probably guess where the actual meaningful part is). As repeaters are invisible, I think it wouldn't even be possible for the nodes around me to ignore all messages from my repeater.

Is this correct or am I missing something?

Jorropo, 2024-11-26:
@xnyhps this has nothing to do with admin channel and affects everything as you correctly pointed out.

The idea I had is that if you randomly flipped bits there would presumably be a carbon based intelligence that would understand a glitchy hopefully not valid UTF8 message to be something wrong while a silicon based intelligence wouldn't.

Jorropo, 2024-11-26:
I've tried fixing it by adding an AEAD channel mode option but changing the architecture of the C++ code to integrate an extra option, do it correctly on all targets, was not fun so I gave up.
If someone care about fixing this it is worth reopening, I don't really.

garthvh, 2025-06-20:
@pqcfox done good luck!

david415, 2025-07-27:
> [@pqcfox](https://github.com/pqcfox) done good luck!

don't be an ass

david415, 2025-07-27:
> Hi! I'm personally very interested in implementing this as a feature, and can take on the workload--as an applied cryptographer having an AEAD channel mode would be important to me.
> 
> Would it be possible to reopen this issue, just so the work I'm doing is going toward an existing issue? Thanks!

Food for thought:

I am so embarrassed for Meshtastic because there are many MANY definitions of a bad person, a person who causes harm to others or who through their inaction allows harm to come to others, however in today's day an age there is also a kind of cryptographic / information theoretic definitions of a "bad person" which certainly is a person who releases insecure cryptographic protocols that could allow for human rights violations.

This is not a game. People are dying in the streets. The secret police in my country are acquiring more and more power. The people need strong encryption.

khimaros, 2025-11-28:
is this issue resolved?

Jorropo, 2025-11-28:
No, the bot is annoying here.

pqcfox, 2026-02-24:
Hiya! Thanks for the tag ^^

I would suggest that this potentially could use some careful modeling out-of-band. In particular, from my past review of this issue, here's some relevant concerns:

- The use of an 8-byte auth tag lower collision resistance to 32 bits, which depending on the use case may not be advisable and could warrant a review of the protocol as a whole

- User-configurable cryptography not aimed at cryptographic engineers can often ends in disaster; the AEAD option could be suited as a one-time feature flag, but having many variants of a protocol in flight makes analysis and later protocol formalization much harder

- In particular, downgrading attacks might be relevant if receiving CTR messages causes the rest of an exchange to be unauthed

- The CCM implementation should probably be run through Wycheproof, as it's (to my knowledge) not presently audited or tested for correctness (though IIRC the external crypto libs used aren't audited either)

In particular, I do think as a matter of shared responsibility that any additional claim of cryptographic guarantees (e.g. authenticated channels) needs to be paired with a lot of attention to these kinds of underlying details which might undermine the claimed property. 

(This of course can be noted as experimental work, but I'd be remiss if I didn't wave my arms a bit and say that there are possible risks which need to be mitigated before one can responsibly claim to users that they can rely on a new property; some of these are less concerns regarding the suggested strategy as much as what I understand to be proximate needs that could use attention in conjunction with this issue.)

Avamander, 2026-03-03:
I think this proposal needs careful review before anything is implemented. There are quite a few pitfalls with CCM usage in general and in implementing it. Rolling out a cryptosystem, especially something for messaging is _very_ difficult to do correctly. Mistakes are even more annoying to remedy.

garthvh, 2026-09-13:
Adding a set of concrete 2.8-era targets for this same primitive, from the "meshtastic-is-dead" talk (Crypto attacks, "Tamper without the key").

The root cause is the one this issue already describes, AES-CTR with no MAC, and the title already covers the key point that the PSK isn't needed. What the talk adds is that for certain fields you don't need to reconstruct the plaintext at all. This issue's worked example builds `bitstream = cypherText ^ knownPlaintext` to forge chosen content, which needs a good guess at the original message. But a single bit flipped at a known offset has a deterministic effect on the plaintext regardless of what the rest of the message says, and several fields worth attacking now sit at predictable offsets:

- **`bitfield`**, at a fixed offset from the end of the encoded `Data`. Flipping one bit sets `want_response` (every receiver that decrypts it generates a reply, an amplification lever) or `OK_TO_MQTT`. The latter is the interesting one: it pushes traffic to public MQTT against the sender's explicit choice, so an attacker with no key for the channel can cause its contents to be uplinked to the internet where a gateway that does hold the key will publish them. The shifts are public constants, `BITFIELD_OK_TO_MQTT_SHIFT` and `BITFIELD_WANT_RESPONSE_SHIFT`, applied in `perhapsEncode()` (`src/mesh/Router.cpp`).

- **NodeInfo `public_key`**, 32 bytes near the end of the message. XORing garbage over it needs no knowledge of the original and corrupts the key a receiver binds to that node number under TOFU, a targeted denial of PKI messaging with that peer. Substituting a chosen key requires knowing the original bytes, but those are broadcast in the clear in other NodeInfo packets, so that precondition is often satisfiable.

- **`id`** is a plaintext header field, not ciphertext, so rewriting it needs no crypto trick at all. This issue notes in passing that a replay may be dropped as a duplicate; rewriting `id` is the general answer to that, since dedup keys on it.

None of these are separate bugs. They're the same missing-integrity root cause this issue was opened for, and they're consistent with what `SECURITY.md` documents as accepted. Recording them here because the accepted-limitation language is written around *possession of the PSK*, and these particular edits sit outside that framing: the tamper primitive needs no key, only a known offset. Worth weighing when the AEAD discussion above gets revisited.

2.8's XEdDSA signing mitigates part of this for broadcast traffic, since a receiver that has seen a node sign can reject unsigned packets claiming that identity, and a bit-flipped packet fails signature verification. It does not cover unicast or unsigned-capable paths.

matutetandil, 2026-09-14:
@garthvh - AEAD landed in develop this morning as d05fbec64 (#9749), so on channels with `use_aead` set the primitive your first two targets rely on is closed.

The `bitfield` and the NodeInfo `public_key` both sit inside the encoded `Data`, which is what gets encrypted, so the CCM tag covers them: a bit flipped anywhere in the ciphertext fails authentication and the packet is dropped instead of decoded. `from` and `to` are passed as associated data as well, so the destination can't be rewritten either. The hop fields are deliberately excluded, since relays legitimately rewrite them.

Your third point still stands. `id` is a plaintext header field rather than ciphertext, so nothing here touches it. That's the same thread as the Store & Forward problem @caveman99 wrote up on the PR: `preparePayload()` preserves `from` and `id` while rewriting `to`, so the replay reuses (key, nonce). The fix is a fresh id for the replay, but the phone apps dedup and thread S&F history on that id, so it needs its own PR with app-side coordination.

Worth being clear that AEAD is opt-in and off by default, so the default configuration is unchanged and everything this issue documents still applies there.