meshtastic/firmware #4038: [Bug]:

slash-bit, 2024-06-04:
### Category

Other

### Hardware

T-Echo

### Firmware Version

2.3.11

### Description

I wanted to make a small contribution in relation to Screen Frame reload issue ( may be not such a big one) , but every time setFrame run the page reverts back to Message or Module frame , but not the same frame that was viewed by user before, which could be annoying having to manually switch back to say Status frame.
Here is the change needed, please have a look and comment if needed. I am happy to submit PR if all OK. 
There are some caveats: After Resetting NodeDB or enabling / disabling a Module , the frame will not necessarily revert back to originally viewed frame. But after the device rebooted and there are more than 5 nodes in the list , it is quite stable.
Tested on T-Echo.
https://github.com/meshtastic/firmware/compare/master...meshi-bit:meshtastic-echo-echo:screen-frames

todd-herbert, 2024-06-04:
Did we speak about this on the Discord server they other day? Would you be happy for me to sort out some of those caveats?

slash-bit, 2024-06-04:
Hey, yes it was me on discord :-) After some thinking , I though this OK to implement. But I am more than happy take help with getting rid off those caveats. Thanks ! :-)

todd-herbert, 2024-06-04:
Hey if you're happy with it, go for it! I might add some extras later on / tidy some E-Ink code that this replaces 👍

caveman99, 2024-06-05:
This needs a bit more polish, but it's a good start. If you draft a PR we can comment and build on it (e.g. what if framenum-2 doesn't exist as well?) :-)

slash-bit, 2024-06-05:
Just saw the comments and created a PR as Draft now.
https://github.com/meshtastic/firmware/pull/4043

slash-bit, 2024-06-05:
Yes, the (numframes - 2) is not ideal. I will do some checks, what is the absolute minimum number of frames possible. I believe it should be at least 2.
Or we can revert back to frame 0, if the currentFrameNum happens to be outside of  frames available (numframes)

closed 2024-07-11 by thebentern