meshtastic/firmware #4039: [Bug]: rak4631 no lock on GPS results in shutdown at first position broadcast interval

gunkl, 2024-06-04:
### Category

Other

### Hardware

Rak4631

### Firmware Version

2.3.10

### Description

2.3.10 and probably 2.3.6 on rak4631 with gps module
With GPS enabled, but no lock, it appears the GPS reports a lat/lon of 0/0 and then the system indicates "long button press shutdown" at the time where it hits the broadcast interval (15 mins)
See attached screen shot of serial log. I was able to replicate this easily and repeatably.

Disabling Smart Position has no effect
Disabling GPS and indicating fixed position resolves the shutdown issue

This might also manifest itself with GPS enabled, consider the scenario where you have GPS lock for a period of time and transmit is OK, but for some reason GPS loses lock, this may repeat. I'm pretty sure this happened to me at least once whereby the unit shutdown in the middle of the night unattended.

Sorry that i did not produce text log copy, but putty disconnects the moment the device shuts down, and prevents me from selecting the log text. I could have made it save to log. Derp.

[image]

ianmcorvidae, 2024-06-04:
if you've got the GPS in slot D, that's your issue. The fix is to either change the "user button" pin to something other than the default (pin 9), which is part of the connector on slot D, or move the GPS to slot A as recommended by the docs.

clwgh, 2024-06-05:
This looks like the same issue [previously reported](https://meshtastic.discourse.group/t/rak-4631-shutting-down-without-button-push/12186) with the RAK shutting down at 15 minutes. The button workaround at the end of that discussion was used in a [caving project](https://github.com/semper-ad-fundum/vangelis).

> `device.button_gpio 33` — fixes an issue where the board believes barometric sensor sends a long button press and shuts down

closed 2024-06-05 by caveman99