meshtastic/firmware #4040: [Bug]: Telemetry Module Issues BME680 on Heltec V3

HarukiToreda, 2024-06-05:
### Category

Other

### Hardware

Heltec V3

### Firmware Version

2.3.11

### Description

Couple new issues but seems to be closer to working how it used to. This is a BME680 sensor on a Heltec V3. 

1. Environmental data is not initializing and showing on the screen after boot. It only shows once after 15min default settings. I changed the default to 5min and it showed after 5min. But before, it would show as soon as it took the first measurement. Waiting 15min after boot. See logs below for when I changed it to 5min. 
```
INFO | ??:??:?? 7 [Screen] Done with boot screen...
DEBUG | ??:??:?? 7 [Screen] showing standard frames
DEBUG | ??:??:?? 7 [Screen] Module wants a UI Frame
DEBUG | ??:??:?? 7 [Screen] Showing 1 module frames
DEBUG | ??:??:?? 7 [Screen] Total frame count: 104
DEBUG | ??:??:?? 7 [Screen] Added modules. numframes: 1
DEBUG | ??:??:?? 7 [Screen] Finished building frames. numframes: 4
INFO | ??:??:?? 11 [EnvironmentTelemetryModule] Environment Telemetry: Initializing
INFO | ??:??:?? 11 [EnvironmentTelemetryModule] BME680 state read from /prefs/bsec.dat.
INFO | ??:??:?? 11 [EnvironmentTelemetryModule] Init sensor: BME680 with the BSEC Library version 2.5.0.2
INFO | ??:??:?? 11 [EnvironmentTelemetryModule] Opened BME680 sensor on i2c bus
INFO | ??:??:?? 125 [PowerFSM] lsEnter begin, ls_secs=300
INFO | ??:??:?? 126 [PowerFSM] setup LORA_DIO1 (GPIO14) with wakeup by external source
INFO | ??:??:?? 156 [PowerFSM] Exit light sleep cause: 4
INFO | ??:??:?? 156 [PowerFSM] setup LORA_DIO1 (GPIO14) with wakeup by external source
INFO | ??:??:?? 156 [PowerFSM] Exit light sleep cause: 4
DEBUG | ??:??:?? 156 [Power] Battery: usbPower=1, isCharging=1, batMv=4235, batPct=100
WARN | ??:??:?? 156 [EnvironmentTelemetryModule] runTrigger BSEC2 code: 100
INFO | ??:??:?? 156 [PowerFSM] setup LORA_DIO1 (GPIO14) with wakeup by external source
INFO | ??:??:?? 186 [PowerFSM] Exit light sleep cause: 4
INFO | ??:??:?? 186 [PowerFSM] setup LORA_DIO1 (GPIO14) with wakeup by external source
INFO | ??:??:?? 186 [PowerFSM] Exit light sleep cause: 4
DEBUG | ??:??:?? 186 [Power] Battery: usbPower=1, isCharging=1, batMv=4240, batPct=100
WARN | ??:??:?? 186 [EnvironmentTelemetryModule] runTrigger BSEC2 code: 100
INFO | ??:??:?? 186 [PowerFSM] setup LORA_DIO1 (GPIO14) with wakeup by external source
INFO | ??:??:?? 216 [PowerFSM] Exit light sleep cause: 4
INFO | ??:??:?? 216 [PowerFSM] setup LORA_DIO1 (GPIO14) with wakeup by external source
INFO | ??:??:?? 216 [PowerFSM] Exit light sleep cause: 4
DEBUG | ??:??:?? 216 [Power] Battery: usbPower=1, isCharging=1, batMv=4237, batPct=100
WARN | ??:??:?? 216 [EnvironmentTelemetryModule] runTrigger BSEC2 code: 100
INFO | ??:??:?? 216 [PowerFSM] setup LORA_DIO1 (GPIO14) with wakeup by external source
INFO | ??:??:?? 246 [PowerFSM] Exit light sleep cause: 4
INFO | ??:??:?? 246 [PowerFSM] setup LORA_DIO1 (GPIO14) with wakeup by external source
INFO | ??:??:?? 246 [PowerFSM] Exit light sleep cause: 4
DEBUG | ??:??:?? 246 [Power] Battery: usbPower=0, isCharging=0, batMv=4200, batPct=100
WARN | ??:??:?? 246 [EnvironmentTelemetryModule] runTrigger BSEC2 code: 100
INFO | ??:??:?? 247 [PowerFSM] setup LORA_DIO1 (GPIO14) with wakeup by external source
INFO | ??:??:?? 277 [PowerFSM] Exit light sleep cause: 4
INFO | ??:??:?? 277 [PowerFSM] setup LORA_DIO1 (GPIO14) with wakeup by external source
INFO | ??:??:?? 277 [PowerFSM] Exit light sleep cause: 4
DEBUG | ??:??:?? 277 [Power] Battery: usbPower=1, isCharging=1, batMv=4225, batPct=100
WARN | ??:??:?? 277 [EnvironmentTelemetryModule] runTrigger BSEC2 code: 100
INFO | ??:??:?? 277 [PowerFSM] setup LORA_DIO1 (GPIO14) with wakeup by external source
INFO | ??:??:?? 307 [PowerFSM] Exit light sleep cause: 4
INFO | ??:??:?? 307 [PowerFSM] setup LORA_DIO1 (GPIO14) with wakeup by external source
INFO | ??:??:?? 307 [PowerFSM] Exit light sleep cause: 4
DEBUG | ??:??:?? 307 [Power] Battery: usbPower=1, isCharging=1, batMv=4224, batPct=100
WARN | ??:??:?? 307 [EnvironmentTelemetryModule] runTrigger BSEC2 code: 100
INFO | ??:??:?? 307 [PowerFSM] setup LORA_DIO1 (GPIO14) with wakeup by external source
INFO | ??:??:?? 321 [PowerFSM] Exit light sleep gpio: btn=1
INFO | ??:??:?? 321 [PowerFSM] Exit state: LS
DEBUG | ??:??:?? 321 [PowerFSM] Enter state: ON
INFO | ??:??:?? 321 [Screen] Turning on screen
WARN | ??:??:?? 321 [EnvironmentTelemetryModule] runTrigger BSEC2 code: 100
DEBUG | ??:??:?? 321 [EnvironmentTelemetryModule] BME680 not updated, IAQ accuracy is 0 < 2
INFO | ??:??:?? 321 [EnvironmentTelemetryModule] (Sending): barometric_pressure=10.124850, current=0.000000, gas_resistance=228.841843, relative_humidity=40.076012, temperature=28.481543
INFO | ??:??:?? 321 [EnvironmentTelemetryModule] (Sending): voltage=1.596691, IAQ=66, distance=1.614969, lux=1.614965
INFO | ??:??:?? 321 [EnvironmentTelemetryModule] Sending packet to mesh
```
2. The Enviromental screen is showing voltage /current even though setting for it is turned off. This makes the screen cut off the IAQ data.

[image]

HarukiToreda, 2024-06-05:
If a phone is connected, it shows on the screen pretty quick. Not sure what would cause that.

```
DEBUG | ??:??:?? 47 [RadioIf] Starting low level send (id=0x4ed8dde2 fr=0xcc to=0xff, WantAck=0, HopLim=3 Ch=0x8 encrypted hopStart=3 priority=10)
DEBUG | ??:??:?? 47 [RadioIf] (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=46, time 575 ms
DEBUG | ??:??:?? 47 [RadioIf] AirTime - Packet transmitted : 575ms
INFO  | ??:??:?? 47 From Radio onread
INFO  | ??:??:?? 47 getFromRadio=STATE_SEND_PACKETS
DEBUG | ??:??:?? 47 encoding toPhone packet to phone variant=11, 12 bytes
INFO  | ??:??:?? 47 From Radio onread
INFO  | ??:??:?? 47 getFromRadio=STATE_SEND_PACKETS
DEBUG | ??:??:?? 47 phone downloaded packet (id=0x4ed8dde2 fr=0xcc to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=67 hopStart=3 priority=10)
DEBUG | ??:??:?? 47 encoding toPhone packet to phone variant=2, 55 bytes
INFO  | ??:??:?? 47 From Radio onread
DEBUG | ??:??:?? 47 [RadioIf] Completed sending (id=0x4ed8dde2 fr=0xcc to=0xff, WantAck=0, HopLim=3 Ch=0x8 encrypted hopStart=3 priority=10)
DEBUG | ??:??:?? 62 [Power] Battery: usbPower=1, isCharging=1, batMv=4239, batPct=100
INFO  | ??:??:?? 65 [PowerFSM] Turning off screen
DEBUG | ??:??:?? 72 [Button] Enter state: ON
INFO  | ??:??:?? 72 [Screen] Turning on screen
DEBUG | ??:??:?? 73 [EnvironmentTelemetryModule] BME680 not updated, IAQ accuracy is 0 < 2
INFO  | ??:??:?? 73 [EnvironmentTelemetryModule] (Sending): barometric_pressure=10.129648, current=-0.500000, gas_resistance=190.581741, relative_humidity=51.260315, temperature=27.410236
INFO  | ??:??:?? 73 [EnvironmentTelemetryModule] (Sending): voltage=-0.200000, IAQ=68, distance=1.614986, lux=1.579515
INFO  | ??:??:?? 73 [EnvironmentTelemetryModule] Sending packet to phone
INFO  | ??:??:?? 73 Telling client we have new packets 5
INFO  | ??:??:?? 73 BLE notify fromNum
INFO  | ??:??:?? 73 From Radio onread
INFO  | ??:??:?? 73 getFromRadio=STATE_SEND_PACKETS
DEBUG | ??:??:?? 73 phone downloaded packet (id=0x4ed8dde3 fr=0xcc to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=67 priority=10)
DEBUG | ??:??:?? 73 encoding toPhone packet to phone variant=2, 81 bytes
INFO  | ??:??:?? 73 From Radio onread
DEBUG | ??:??:?? 82 [Power] Battery: usbPower=1, isCharging=1, batMv=4255, batPct=100
```

HarukiToreda, 2024-06-05:
Don't know what happened but after loading it to my other nodes, it worked as it did before. Closing this issue

caveman99, 2024-06-05:
Telemetry overall is taking the first measurement only after the sensor interval has passed. I noticed that too. As for the 'ghost values' - can you check the latest nightnly/master firmware? we pulled in a potential fix yesterday.

HarukiToreda, 2024-06-05:
> Telemetry overall is taking the first measurement only after the sensor interval has passed. I noticed that too. As for the 'ghost values' - can you check the latest nightnly/master firmware? we pulled in a potential fix yesterday.

yes, that fixed the ghost values issue. Still taking a while till the value shows on the screen after reboot. It shows fast when a phone is linked to it though.

closed 2024-06-17 by HarukiToreda