meshtastic/firmware #4045: [Bug]: Altitude problems in Android app and in MQTT

TheDomanNow, 2024-06-05:
### Category

Other

### Hardware

Heltec V3

### Firmware Version

2.3.10.d19607b

### Description

Hi,

[ANDROID APP > SETTINGS > POSITION CONFIG > Use fixed position]
1. When fixed altitude is set, after device is restarted after some time, the value is being reset to "0", not always but it happens form time to time.
2. MQTT is not passing the altitude value (can see "0" altitude via MQTT).

Heltec V3.1 | Meshtastic Firmware 2.3.10.d19607b Beta | Android app + BLE/IP

BR, Doman.

madeofstown, 2024-06-06:
What is your GPS Mode setting? I've also struggled to get the GPS position to stick after power cycling on a Pico. What eventually worked for me was to allow the position from my phone to be captured by the node. Then I disable the setting to allow GPS from the phone before reconnecting to the node. After another restart I found that the coordinates hold.

TheDomanNow, 2024-06-06:
2x ENABLED + Fixed position.

Heltec V3 doesn't have GPS. Devices are not connected to smartphone.

madeofstown, 2024-06-06:
> Devices are not connected to smartphone.

So I'm assuming that your Android device doesn't have a GPS... 
Have you tried changing the mode setting to "Not Present"? With the setting on "Enabled" I think the device may be looking for a GPS that isn't available could be overwriting your manual entry with empty values.