meshtastic/firmware #4054: [Bug]:

pepefrog1234, 2024-06-08:
### Category

Hardware Compatibility, Other

### Hardware

Other

### Firmware Version

CDEBYTE_EORA_S3

### Description

The warning message “[RadioIf] Can not send yet, busyRx” kept appearing in the log file, and the message was received by the V3 beside me when it could not be sent successfully.

### Relevant log output

```Shell
INFO | ??:??:?? 5 Booted, wake cause 0 (boot count 1), reset_reason=reset

DEBUG | ??:??:?? 5 Filesystem files (495616/1048576 Bytes):

DEBUG | ??:??:?? 5 /prefs/channels.proto (216 Bytes)

DEBUG | ??:??:?? 5 /prefs/config.proto (98 Bytes)

DEBUG | ??:??:?? 5 /prefs/db.proto (1938 Bytes)

DEBUG | ??:??:?? 5 /prefs/module.proto (104 Bytes)

DEBUG | ??:??:?? 5 /static/.gitkeep (0 Bytes)

DEBUG | ??:??:?? 5 /static/Logo_Black.svg.gz (602 Bytes)

DEBUG | ??:??:?? 5 /static/Logo_White.svg.gz (610 Bytes)

DEBUG | ??:??:?? 5 /static/apple-touch-icon.png.gz (3164 Bytes)

DEBUG | ??:??:?? 5 /static/favicon.ico.gz (2270 Bytes)

DEBUG | ??:??:?? 5 /static/icon.svg.gz (852 Bytes)

DEBUG | ??:??:?? 5 /static/index-BMhLjTmL.css.gz (16270 Bytes)

DEBUG | ??:??:?? 5 /static/index-Bt2MYVVB.js.gz (203021 Bytes)

DEBUG | ??:??:?? 5 /static/index-CGqDWSD_.js.gz (348 Bytes)

DEBUG | ??:??:?? 6 /static/index.html.gz (548 Bytes)

DEBUG | ??:??:?? 6 /static/maplibre-gl-BXZZhNda.js.gz (210451 Bytes)

DEBUG | ??:??:?? 6 /static/robots.txt.gz (42 Bytes)

DEBUG | ??:??:?? 6 /static/site.webmanifest.gz (197 Bytes)

DEBUG | ??:??:?? 6 Using analog input 1 for battery level

INFO | ??:??:?? 6 ADCmod: ADC Characterization based on Two Point values and fitting curve coefficients stored in eFuse

INFO | ??:??:?? 6 Scanning for i2c devices...

DEBUG | ??:??:?? 6 Scanning for i2c devices on port 2

DEBUG | ??:??:?? 6 Scanning for i2c devices on port 1

DEBUG | ??:??:?? 6 I2C device found at address 0x3c

INFO | ??:??:?? 6 ssd1306 display found
[... 1356 lines cut ...]

DEBUG | ??:??:?? 472 [DeviceTelemetryModule] updateTelemetry LOCAL

DEBUG | ??:??:?? 472 [DeviceTelemetryModule] Node status update: 26 online, 26 total

INFO | ??:??:?? 472 [DeviceTelemetryModule] Sending packet to phone

INFO | ??:??:?? 472 Telling client we have new packets 19

INFO | ??:??:?? 472 BLE notify fromNum

INFO | ??:??:?? 472 From Radio onread

INFO | ??:??:?? 472 getFromRadio=STATE_SEND_PACKETS

DEBUG | ??:??:?? 472 phone downloaded packet (id=0x05abd2ae fr=0x80 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=67 priority=10)

DEBUG | ??:??:?? 472 encoding toPhone packet to phone variant=2, 49 bytes

INFO | ??:??:?? 472 From Radio onread

DEBUG | ??:??:?? 487 [Power] Battery: usbPower=1, isCharging=1, batMv=4766, batPct=100

DEBUG | ??:??:?? 507 [Power] Battery: usbPower=1, isCharging=1, batMv=4728, batPct=100
```

GUVWAF, 2024-06-08:
I'm not sure what the bug is here? That warning is normal, it's either receiving a real packet, so it cannot send yet, or it had a false preamble detection, which it discards after a short while. Then it will send as normal, like you said happened.

pepefrog1234, 2024-06-08:
My test environment has three devices which are: 1. v3 mqtt network node 2. v3 portable device 3. CDEBYTE_EORA_S3 problematic device.
The channel is idle, I send a message but the app application shows that the message cannot be sent and the message [Can not send yet, busyRx] appears in the log.
I tried to send the same message with device #2 (v3) and it sent successfully.

I apologize for not being clear earlier, I've reworked my issue.

GUVWAF, 2024-06-08:
So from which device is the log output? That looks normal.

pepefrog1234, 2024-06-08:
The logs are exported from CDEBYTE_EORA_S3.

GUVWAF, 2024-06-08:
But that logs shows it successfully sent the packets and even another node is rebroadcasting for you.
Which app are you using that shows it cannot be sent?

pepefrog1234, 2024-06-08:
android app.
I have observed that if the message is only one or two characters long, it is more likely to be sent successfully, but if the characters are too long (more than 50 bytes), it will fail to be sent.

GUVWAF, 2024-06-08:
How far are the nodes from each other? If they're in the same room, they might get overloaded.

Can you show a log when the message fails? Preferably from both sending and receiving device.

enforcer2000, 2024-08-09:
I am having the same problem - I have 3 nodes - a PICO, and two Heltec V3s. the PICO and one Heltec are performing fine - the firmware was consistent on v2.3.15.deb7c27 between the two Heltec nodes but only one Heltec and the PICO were communicating. With direct messaging between the Heltecs, the working Heltec was saying the message sent successfully but it was never shown as received in the app (android) on the Heltec I am having issues with. I have just re-flashed the firmware on the problematic Heltec with the latest Alpha release 2.4.2.5b45303 and I am still getting swamped with the following logs as you can see from the timestamps. Even the beacons don't seem to be getting through as the node is not showing up on either the PICO (newly reflashed with flash_nuke) or updating the last seen on the working Heltec...

WARN | 17:12:37 34 [RadioIf] Can not send yet, busyRx
WARN | 17:12:37 34 [RadioIf] Can not send yet, busyRx
WARN | 17:12:37 34 [RadioIf] Can not send yet, busyRx
WARN | 17:12:37 34 [RadioIf] Can not send yet, busyRx
WARN | 17:12:38 34 [RadioIf] Can not send yet, busyRx
WARN | 17:12:38 34 [RadioIf] Can not send yet, busyRx
WARN | 17:12:38 34 [RadioIf] Can not send yet, busyRx
WARN | 17:12:38 34 [RadioIf] Can not send yet, busyRx
WARN | 17:12:38 34 [RadioIf] Can not send yet, busyRx
WARN | 17:12:38 34 [RadioIf] Can not send yet, busyRx
WARN | 17:12:38 35 [RadioIf] Can not send yet, busyRx
WARN | 17:12:38 35 [RadioIf] Can not send yet, busyRx
WARN | 17:12:38 35 [RadioIf] Can not send yet, busyRx
WARN | 17:12:38 35 [RadioIf] Can not send yet, busyRx
WARN | 17:12:38 35 [RadioIf] Can not send yet, busyRx
WARN | 17:12:39 35 [RadioIf] Can not send yet, busyRx
WARN | 17:12:39 35 [RadioIf] Can not send yet, busyRx
WARN | 17:12:39 35 [RadioIf] Can not send yet, busyRx
WARN | 17:12:39 35 [RadioIf] Can not send yet, busyRx
WARN | 17:12:39 35 [RadioIf] Can not send yet, busyRx
WARN | 17:12:39 35 [RadioIf] Can not send yet, busyRx
WARN | 17:12:39 35 [RadioIf] Can not send yet, busyRx
WARN | 17:12:39 36 [RadioIf] Can not send yet, busyRx
WARN | 17:12:39 36 [RadioIf] Can not send yet, busyRx
WARN | 17:12:39 36 [RadioIf] Can not send yet, busyRx
WARN | 17:12:39 36 [RadioIf] Can not send yet, busyRx
WARN | 17:12:39 36 [RadioIf] Can not send yet, busyRx
WARN | 17:12:40 36 [RadioIf] Can not send yet, busyRx
WARN | 17:12:40 36 [RadioIf] Can not send yet, busyRx
WARN | 17:12:40 36 [RadioIf] Can not send yet, busyRx
WARN | 17:12:40 36 [RadioIf] Can not send yet, busyRx
WARN | 17:12:40 36 [RadioIf] Can not send yet, busyRx
WARN | 17:12:40 36 [RadioIf] Can not send yet, busyRx
WARN | 17:12:40 36 [RadioIf] Can not send yet, busyRx
WARN | 17:12:40 36 [RadioIf] Can not send yet, busyRx
WARN | 17:12:40 36 [RadioIf] Can not send yet, busyRx
WARN | 17:12:40 37 [RadioIf] Can not send yet, busyRx
WARN | 17:12:40 37 [RadioIf] Can not send yet, busyRx
WARN | 17:12:40 37 [RadioIf] Can not send yet, busyRx
WARN | 17:12:41 37 [RadioIf] Can not send yet, busyRx
WARN | 17:12:41 37 [RadioIf] Can not send yet, busyRx
WARN | 17:12:41 37 [RadioIf] Can not send yet, busyRx
WARN | 17:12:41 37 [RadioIf] Can not send yet, busyRx
WARN | 17:12:41 37 [RadioIf] Can not send yet, busyRx
WARN | 17:12:41 37 [RadioIf] Can not send yet, busyRx
WARN | 17:12:41 37 [RadioIf] Can not send yet, busyRx
WARN | 17:12:41 37 [RadioIf] Can not send yet, busyRx
WARN | 17:12:41 37 [RadioIf] Can not send yet, busyRx
WARN | 17:12:41 37 [RadioIf] Can not send yet, busyRx
WARN | 17:12:41 37 [RadioIf] Can not send yet, busyRx
WARN | 17:12:41 37 [RadioIf] Can not send yet, busyRx
WARN | 17:12:41 38 [RadioIf] Can not send yet, busyRx
WARN | 17:12:41 38 [RadioIf] Can not send yet, busyRx
WARN | 17:12:41 38 [RadioIf] Can not send yet, busyRx
WARN | 17:12:41 38 [RadioIf] Can not send yet, busyRx
WARN | 17:12:41 38 [RadioIf] Can not send yet, busyRx
WARN | 17:12:42 38 [RadioIf] Can not send yet, busyRx
WARN | 17:12:42 38 [RadioIf] Can not send yet, busyRx
WARN | 17:12:42 38 [RadioIf] Can not send yet, busyRx
WARN | 17:12:42 38 [RadioIf] Can not send yet, busyRx
WARN | 17:12:42 38 [RadioIf] Can not send yet, busyRx
WARN | 17:12:42 38 [RadioIf] Can not send yet, busyRx
WARN | 17:12:42 38 [RadioIf] Can not send yet, busyRx
WARN | 17:12:42 38 [RadioIf] Can not send yet, busyRx
WARN | 17:12:42 39 [RadioIf] Can not send yet, busyRx
WARN | 17:12:42 39 [RadioIf] Can not send yet, busyRx
WARN | 17:12:43 39 [RadioIf] Can not send yet, busyRx
WARN | 17:12:43 39 [RadioIf] Can not send yet, busyRx
[... 9 lines cut ...]
WARN | 17:12:43 40 [RadioIf] Can not send yet, busyRx
WARN | 17:12:43 40 [RadioIf] Can not send yet, busyRx
WARN | 17:12:43 40 [RadioIf] Can not send yet, busyRx
WARN | 17:12:43 40 [RadioIf] Can not send yet, busyRx
WARN | 17:12:43 40 [RadioIf] Can not send yet, busyRx
WARN | 17:12:43 40 [RadioIf] Can not send yet, busyRx
WARN | 17:12:44 40 [RadioIf] Can not send yet, busyRx
WARN | 17:12:44 40 [RadioIf] Can not send yet, busyRx
WARN | 17:12:44 40 [RadioIf] Can not send yet, busyRx
WARN | 17:12:44 40 [RadioIf] Can not send yet, busyRx
WARN | 17:12:44 40 [RadioIf] Can not send yet, busyRx
WARN | 17:12:44 40 [RadioIf] Can not send yet, busyRx
WARN | 17:12:44 40 [RadioIf] Can not send yet, busyRx
WARN | 17:12:44 40 [RadioIf] Can not send yet, busyRx
WARN | 17:12:44 40 [RadioIf] Can not send yet, busyRx
WARN | 17:12:44 41 [RadioIf] Can not send yet, busyRx
WARN | 17:12:44 41 [RadioIf] Can not send yet, busyRx
WARN | 17:12:44 41 [RadioIf] Can not send yet, busyRx
WARN | 17:12:44 41 [RadioIf] Can not send yet, busyRx
WARN | 17:12:44 41 [RadioIf] Can not send yet, busyRx
WARN | 17:12:44 41 [RadioIf] Can not send yet, busyRx
WARN | 17:12:44 41 [RadioIf] Can not send yet, busyRx
WARN | 17:12:45 41 [RadioIf] Can not send yet, busyRx
WARN | 17:12:45 41 [RadioIf] Can not send yet, busyRx
WARN | 17:12:45 41 [RadioIf] Can not send yet, busyRx

GerritS76, 2024-08-13:
Same issue here, started from 2.4.0 on both Heltec V3 as well Heltec Wireless Stick Lite v3.

Also noticed none of the devices are connecting to MQTT Brokers..

garthvh, 2024-08-13:
Turn of mqtt and debug, those are normal errors if you are congested

GUVWAF, 2024-08-13:
The `[RadioIf] Can not send yet, busyRx` logs are normal indeed, but should stop being printed after the maximum packet length, which is a couple of seconds or so on LongFast. If it's longer than that, then definitely something is wrong. I haven't encountered this on 2.4 myself.

enforcer2000, 2024-08-13:
So I wanted to drop an update on this from my perspective...
After a reflash with " full erase and install" set on the problematic Heltec V3, this seemed to have corrected the issue so there must have been some corrupt configuration somewhere. Unfortunately I lost all my previous beacon info but at least I have it working again.