meshtastic/firmware #4061: [Bug]: uBlox M10 GPS doesn't sleep if device is switched off

craft4tnt, 2024-06-09:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

2.3.12

### Description

UBlox M10x GPS module stays in aquisition mode (~12mA) and does not enter sleep mode (~50uA) if meshtastic device is switched off. 
The receivers power states are controlled by `UBX-RXM-PMREQ` commands. UART_RX activity is defined as a wakeup source. During normal operation, M10 GPS power control works fine. As soon as the meshtastic device is switched off / to deep sleep, the GPS receiver will stay in aquisition mode and drains the battery.

Power consumtion was measured with [Current Ranger](https://lowpowerlab.com/guide/currentranger/)

Will provide PR with fix immediately

GPSFan, 2024-06-09:
The prepareDeepSleep method in GPS.cpp probably could be used to send a PMREQ to the GPS with the duration set to max (12 days). When Meshtastic (esp32) wakes up from DeepSleep it reboots so the GPS will be re-configured to normal operation then, as long as the PMREQ wakeup sources can wake it up.

craft4tnt, 2024-06-09:
The wierd thing is that a UBX-RXM-PMREQ with duration=0 and wakeupSources->uartrx=1 never set the module to sleep even if there are no level changes on the UART line. I tried keeping it pulled high and low, both didn't work. It might be a bug which is specific to the M10, since the PMREQ messages seem deprecated but still supported for the newer models.
I didn't try the maximum duration value of 12 days but it also doesn't really fix the issue.
Thats why I decided to go with the external control since the PIN_GPS_EN signal is already implemented.

GPSFan, 2024-06-09:
What firmware version is your M10? PMREQ is not deprecated (as yet) according to the SPG 5.10 and 6.00 firmware release notes.

craft4tnt, 2024-06-09:
You are right, PMREQ is not deprecated but I get the impression it's not the favoured way of doing power management for the M10 anymore because it is only briefly mentioned in the [Integration manual](https://content.u-blox.com/sites/default/files/documents/SAM-M10Q_IntegrationManual_UBX-22020019.pdf) among all the new features from the CFG-PM group.

PR #4062 (2024-06-12): Fix uBlox M10 GPS doesn't sleep if device is switched off
This PR fixes #4061 

The ublox GPS modules power state is controlled by legacy `UBX-RXM-PMREQ` commands. A defined sleep time and/or a wakeup source (in this case UART_RX activity) is defined in the sleep request message (_message_PMREQ_10 in ubx.h). For an unknown reason, the receiver wakes up/stays awake if the meshtastic device is switched off, even if sleep time is set to 0 (wakeup only by external events).

To fix this issue, the existing signal `PIN_GPS_EN` is used to control to the **EXTINT** pin of the receiver. (PIN_GPS_EN pin [... cut]
closed 2024-06-12 by thebentern