meshtastic/firmware #4065: [Bug]: WiFi MQTT Not Working After Enabling Bluetooth MQTT Proxy on Phone

Oliver0804, 2024-06-10:
### Category

BLE, WiFi

### Hardware

T-Beam

### Firmware Version

2.3.10

### Description

After enabling the Bluetooth MQTT proxy on the phone, switching to WiFi mode causes the WiFi MQTT to stop working.

spqr1234, 2024-06-11:
I have noticed MQTT does not work at all on the Primary Channel 0 from a WiFi node on my Heltec v3 devices.  I am unable to send or receive messages on the Primary channel when using MQTT via the WiFi interface.  It also does not work sending Direct messages either, although does receive Direct messages.  It works on Secondary channels and when using the Bluetooth direct connection with MQTT Proxy.  It seems to me MQTT functionality in the WiFi environment is not correctly implemented

garthvh, 2024-06-11:
You can use a he proxy or WiFi, not both

closed 2024-06-11 by garthvh

spqr1234, 2024-06-11:
Selecting or deselected Proxy whilst in WiFi mode makes no difference to me.  The node is still unable to send/receive MQTT messages on the Primary Channel 0.  Having Proxy selected/deselected in WiFi mode  has no affect on sending/receiving MQTT messages through Secondary channels, these continue to work.

Oliver0804, 2024-06-12:
> You can use a he proxy or WiFi, not both

Hi garthvh, I'm not sure if I'm expressing this correctly.

I've noticed that when the Bluetooth proxy function is enabled, switching to WIFI mode generally allows MQTT to work over WIFI. However, I found that it's unable to connect. I think this might be a firmware issue.

garthvh, 2024-06-12:
The proxy uses internet from the phone and can't be used with wifi

Oliver0804, 2024-06-12:
> The proxy uses internet from the phone and can't be used with wifi

Thank you for your reply.😘

I understand that on an ESP32-based device, Wi-Fi and Bluetooth cannot be used simultaneously. However, my issue is that if the device is in proxy mode (using Bluetooth), and then I configure the device to use Wi-Fi, it causes the ESP32 to not connect to the MQTT server via Wi-Fi, possibly continuously waiting for the Bluetooth proxy. 

I suspect this might be a bug in the firmware. 

Once I confirm how the issue occurs, I will report it to the firmware team.

spqr1234, 2024-06-14:
Oliver0804.. your observation looks very similar to my own.  I created a thread that's been picked up by the developers.  I noticed MQTT does not work via the web GUI when connected via Wifi on Channel 0.. but other Channels worked, and switching to the app and connecting via bluetooth with the proxy enabled overcomes the problem.  You can see my observations here:  
[https://github.com/meshtastic/firmware/issues/4076#issuecomment-2161446851](url)