meshtastic/firmware #4066: [Bug]: heltec-wsl-v3 / can't compile - ButtonThread.cpp.o

roha-github, 2024-06-10:
### Category

Hardware Compatibility

### Hardware

Other

### Firmware Version

2.3.10.d19607b

### Description

I can't compile for target Heltec Wireless Stick V3. Other ESP32 devices don't have problems.

My dev environment ...

VS Code Version: 1.90.0 (user setup)
Commit: 89de5a8d4d6205e5b11647eb6a74844ca23d2573
Date: 2024-06-04T19:33:54.889Z
Electron: 29.4.0
ElectronBuildId: 9593362
Chromium: 122.0.6261.156
Node.js: 20.9.0
V8: 12.2.281.27-electron.0
OS: Windows_NT x64 10.0.22631
PlatformIO: Core 6.1.15·Home 3.4.4

```
C:\Users\Developer\.platformio\penv\Scripts\platformio.exe run --environment heltec-wsl-v3 

Processing heltec-wsl-v3 (board: heltec_wifi_lora_32_V3; platform: platformio/espressif32@6.3.2; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/heltec_wifi_lora_32_V3.html
PLATFORM: Espressif 32 (6.3.2) > Heltec WiFi LoRa 32 (V3)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20009.0 (2.0.9) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
[nanopb] Installing Protocol Buffers dependencies
Requirement already satisfied: protobuf>=3.19.1 in c:\users\rohadeveloper\.platformio\penv\lib\site-packages (4.25.3)
[nanopb] No generation needed.
Found 76 compatible libraries
Scanning dependencies...
Dependency Graph
|-- RadioLib @ 6.5.0
|-- ESP8266 and ESP32 OLED driver for SSD1306 displays @ 4.3.0+sha.ee628ee
|-- OneButton @ 2.5.0
|-- arduino-fsm @ 2.2.0+sha.7db3702
|-- TinyGPSPlus @ 1.0.3+sha.71a82db
|-- Thread @ 0.0.0+sha.1ae8778
|-- Nanopb @ 0.4.8
|-- ErriezCRC32 @ 1.0.1
|-- NonBlockingRTTTL @ 1.3.0
|-- SparkFun ATECCX08a Arduino Library @ 1.3.1+sha.5cf62b3
|-- PubSubClient @ 2.8.0
|-- NTPClient @ 3.2.1
|-- Syslog @ 2.0.0
|-- Adafruit BusIO @ 1.16.1
|-- Adafruit Unified Sensor @ 1.1.14
|-- Adafruit BMP280 Library @ 2.6.8
|-- Adafruit BMP085 Library @ 1.2.4
|-- Adafruit BME280 Library @ 2.2.4
|-- bsec2 @ 1.7.2502+sha.4a08fd0
|-- BME68x Sensor library @ 1.2.40408
|-- Adafruit MCP9808 Library @ 2.0.2
|-- INA3221 @ 1.0.0+sha.a98160e
|-- Adafruit INA260 Library @ 1.5.2
|-- Adafruit INA219 @ 1.2.3
|-- Adafruit SHTC3 Library @ 1.0.1
|-- Adafruit LPS2X @ 2.0.6
|-- Adafruit SHT31 Library @ 2.2.2
|-- Adafruit PM25 AQI Sensor @ 1.1.0
|-- Adafruit MPU6050 @ 2.2.6
|-- Adafruit LIS3DH @ 1.2.6
|-- SensorLib @ 0.2.0
|-- Adafruit LSM6DS @ 4.7.3
|-- QMC5883LCompass @ 1.2.3
|-- Adafruit VEML7700 Library @ 2.1.6
|-- Adafruit SHT4x Library @ 1.0.4
|-- esp32_https_server @ 1.0.0+sha.23665b3
|-- NimBLE-Arduino @ 1.4.1
|-- libpax @ 1.1.0+sha.7bcd3fc
|-- XPowersLib @ 0.1.4+sha.84b7373
|-- ESP32 Codec2 @ 1.0.1+sha.633326c
|-- SD @ 2.0.0
|-- SPI @ 2.0.0
|-- WiFi @ 2.0.0
|-- Wire @ 2.0.0
|-- Preferences @ 2.0.0
|-- HTTPClient @ 2.0.0
|-- WiFiClientSecure @ 2.0.0
|-- WebServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- LittleFS @ 2.0.0
|-- Adafruit NeoPixel @ 1.12.2
Building in release mode
fatal: not a git repository (or any of the parent directories): .git
Using meshtastic platformio-custom.py, firmware version 2.3.10
Compiling .pio\build\heltec-wsl-v3\src\BluetoothCommon.cpp.o
xtensa-esp32s3-elf-g++: error: CreateProcess: No such file or directory
Compiling .pio\build\heltec-wsl-v3\src\ButtonThread.cpp.o
*** [.pio\build\heltec-wsl-v3\src\BluetoothCommon.cpp.o] Error 1
xtensa-esp32s3-elf-g++: error: CreateProcess: No such file or directory
*** [.pio\build\heltec-wsl-v3\src\ButtonThread.cpp.o] Error 1
=================================================================== [FAILED] Took 73.90 seconds ===================================================================

Environment    Status    Duration
-------------  --------  ------------
heltec-wsl-v3  FAILED    00:01:13.901
============================================================== 1 failed, 0 succeeded in 00:01:13.901 ==============================================================

 *  The terminal process "C:\Users\RoHaDeveloper\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'heltec-wsl-v3'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

roha-github, 2024-06-10:
No issue for 2.3.12.24458a7

caveman99, 2024-06-14:
looks like your toolchain is broken. please remove the .platformio directory and retry. Unfortunately that happens from time to time.

closed 2024-06-14 by caveman99