meshtastic/firmware #4076: [Bug]: MQTT does not work at all on the Primary Channel 0 in WiFi mode

spqr1234, 2024-06-11:
### Category

Other

### Hardware

Heltec V3

### Firmware Version

2.3.11

### Description

With LoRa transmit deactivated, and MQTT activated. MQTT does not work at all on the Primary Channel 0 from a WiFi node on my Heltec v3 devices. I am unable to send or receive messages on the Primary channel when using MQTT via the WiFi interface. It also does not work sending Direct MQTT messages, although does receive Direct messages.  MQTT messages via the WiFi interface work on Secondary channels.  MQTT messages are successfully sent and received via MQTT when the node is connected to the iOS Bluetooth app with the MQTT Proxy.  Selecting/deselecting Proxy and bluetooth has no affect.  Selecting/deselecting Uplink/downlink Channel settings all have no effect on the outcome.  It seems to me there is an issue with MQTT functionality in the WiFi environment with Channel 0.

madeofstown, 2024-06-11:
I believe there was some kind of temporary fix for MQTT that changed how it operates with the disable transmit option. There was a big discussion here before it was implemented but I can't find the link now.

spqr1234, 2024-06-11:
If you do find the link let me know.  LoRa disabled or not, I don't believe messages are being routed via MQTT at all on Channel 0.  I only seem to be able to send/receive LoRa messages over Channel 0 (and mostly local one's at that.. hence the interest in MQTT).  No MQTT traffic seems to appear on the Primary Channel 0.  I do however receive lot's of node updates when setting the Primary Channel 0  to downlink enabled.

spqr1234, 2024-06-11:
Unfortunately setting the Channel 0 to both uplink and downlink doesn't change the situation.  Channel 0 messages are still not being sent or received over MQTT.  By default I usually set to uplink only on Channel 0 in order to avoid downloading 100's of node updates.  As I mentioned earlier, Direct messages are also not being sent with MQTT but... I can receive them, which seems odd.

spqr1234, 2024-06-11:
**To summarise:**

Bluetooth and iOS app MQTT enabled + proxy LoRa disabled: messages sent and received received successfully on Primary Channel 0 with uplink enabled only.
Bluetooth and iOS app MQTT disabled + proxy LoRa enabled: messages sent and received received successfully on Primary Channel 0 with uplink enabled only.
Bluetooth and iOS app MQTT enabled + proxy LoRa enabled: messages sent and received received successfully on Primary Channel 0 with uplink enabled only.

**WiFi Web interface MQTT enabled LoRa disabled: messages sent and received received unsuccessfully on Primary Channel 0 with uplink enabled only.**
WiFi Web interface MQTT disabled LoRa enabled: messages sent and received received successfully on Primary Channel 0 with uplink enabled only.
WiFi Web interface MQTT enabled LoRa enabled: messages sent and received received successfully on Primary Channel 0 with uplink enabled only.

Definitely appears to me that the WiFi implementation of MQTT is not functioning correctly.

madeofstown, 2024-06-11:
What hardware are you testing this on? I will try to reproduce the issue if I have the correct hardware.

madeofstown, 2024-06-11:
I was not able to reproduce your issue, but I am on a v2.3.12.

MY SETUP: Heltec V3. LoRa TX Disabled. WiFi enabled and accessibile over network. (Android lets me connect to IP so I'm not using web interface) MQTT enabled using channel `msh/US/CA/SantaCruz`. Proxy-to-client not enabled. Uplink and Downlink enabled on Primary Channel.

Was able to send a message through the MQTT server that was picked up by my Router Node and rebroadcast over LoRa. I don't have a dummy load antenna to use so I cannot be positive about receiving messages but if you wanna use that same MQTT channel we can send test messages back and forth to confirm.

madeofstown, 2024-06-11:
Edit2: Private messages are also working both ways.

spqr1234, 2024-06-12:
I am using the web interface.  Using the iOS app overcomes the problem.  There seems to be a problem with the web interface.  Can you try connecting to the web interface using a PC and see whether you can replicate.

madeofstown, 2024-06-12:
> I am using the web interface. Using the iOS app overcomes the problem. There seems to be a problem with the web interface. Can you try connecting to the web interface using a PC and see whether you can replicate.
> 
> Edit: root topic on my config is msh/EU_868

I'm working in an area with bad cell reception so my VPN home isn't giving me a strong enough connection to test the web interface at the moment. I'll have to wait until this afternoon when I get home. I was able to see you on the MQTT channel (dingus, lol), but the private message I sent you didn't go through. Just want to confirm that you have upload and download enabled on the primary channel too.

closed 2024-06-13 by garthvh