meshtastic/firmware #4083: [Bug]: ESP32 MT looses last msgs after reboot / feature change

Dorn8010, 2024-06-12:
### Category

Other

### Hardware

Heltec V3

### Firmware Version

2.3.9

### Description

When changing a setting or normal reboot the last messages are lost.
As a suggestion, since the whole purpose of MT are messages, their preservation should have top priority and the last messages should be saved immediately after reception in the NV part of the memory.
So they stay in the device after reboot.

thebentern, 2024-06-12:
Sorry, but we do not have a message store in flash. This is too expensive for most devices where the space we already use is at a premium.

closed 2024-06-12 by thebentern

Dorn8010, 2024-06-12:
But this is not a problem on ESP32 V3 board, if I understand right.
And since the messages are the whole purpose of the effort they are the asset. For any mission, storing the last e.g. 100 msg with a timestamp (if time was received)  and a seq number would make the whole platform more usable and ESP32 V3 has enough space.
Is there any chance this can be added in a form like - if there is enough memory for storage - then store x msgs ?

Dorn8010, 2024-06-12:
For info - I use various T-Beam and mostly Heltec V3 & Wireless paper devices and the limitation for NV memory or flash drive space for text messages don't seem to be a factor in any of them.

Dorn8010, 2024-06-12:
P.S.: And the loss doesn't seem to occur when a shutdown is made and the a reboot.