meshtastic/firmware #4090: [Bug]: Web interface Send/Receive broken

madeofstown, 2024-06-12:
### Category

WiFi, Other

### Hardware

Heltec V3, Heltec Wireless Tracker, Other

### Firmware Version

2.3.12

### Description

This expands https://github.com/meshtastic/firmware/issues/4076.
After testing, I found that the web interface neither sends nor receives messages. Originally thought it was isolated to MQTT but more testing revealed that LoRa is failing as well. 
Devices tested: Heltec V3, Heltec Tracker v1.1, Linux Native (RaspberryPi Zero 2w).

Talie5in, 2024-06-13:
@madeofstown Would this be better located in https://github.com/meshtastic/web for the webclient?

madeofstown, 2024-06-13:
> @madeofstown Would this be better located in https://github.com/meshtastic/web for the webclient?

🤔 you're right