meshtastic/firmware #4096: [Bug]: rak4631 crashes after ~1 day on latest stable fw

robertfoss, 2024-06-13:
### Category

Other

### Hardware

Rak4631

### Firmware Version

2.3.11.2740a56

### Description

Sorry about re-post, but #4073 was made into a discussion after concluding that the version was too old.

After a few days of uptime this device stops responding over bluetooth and is no longer reporting full data on meshmap.

I don't have access to any serial logs due to the long time to reproduce and the device living outside on a roof.

[image]

thebentern, 2024-06-13:
How do you have the node configured?
Do you have another node that you can serial monitor log to capture crash data?

robertfoss, 2024-06-14:
How do I do a remote serial monitor capture?

thebentern, 2024-06-14:
There is no way to do this remotely unfortunately. You'll have to use USB and capture with meshtastic --noproto or a serial terminal like tio or putty

madeofstown, 2024-06-14:
I've been doing battery testing on my RAK node and it has been up and functioning properly for about a week now.

Do you have a lot of nodes in your node list? Are you possibly being flooded with MQTT traffic from the defaulf channel? I've definitely learned my lesson about connecting to the default MQTT channel so maybe that is what's going on here.

madeofstown, 2024-06-20:
I've just now experienced this issue. Upgraded to 2.3.13 and approximately 24 hours later (not sure exactly when because it happened over night) RAK node stops responding and sending telemetry.

robertfoss, 2024-07-01:
I have no battery setup. Running it off a usb power supply.

madeofstown, 2024-07-02:
> Any possibility of posting (or emailing) you battery setup on the RAK?

Yeah, I fried the RAK4631 module. Now everything seems to be working as expected but the batteries only seems to charge every other day or when the percentage drops below 90.  I'm using 3x 150mA 5v rated panels wired in parallel plugged into the solar charging port on the RAK19007 baseboard. I'm using 2x 2600mAh 18650 batteries wired in parallel with a battery charge/discharge protection board between them and the RAK baseboard. 

I plan to eventually install an MPPT charge controller to take over solar charging of the battery and just wire the RAK in parallel it's the battery. Hopefully this will have a more regular charge cycle.

closed 2024-07-02 by garthvh