meshtastic/firmware #4106: [Bug]: Using APP and web client: messages don't show

jranma, 2024-06-14:
### Category

Other

### Hardware

Heltec V3

### Firmware Version

2.3.10.d19607b

### Description

heltec v3 is configured as "ROUTER_CLIENT"
Received/sent messages will show on the paired Android app. 
If I also use the web client, the messages are not displayed in the channels. 
it's a bug because this mode should also allow the gateway to be used as a simple client.

madeofstown, 2024-06-14:
https://github.com/meshtastic/web/issues/228
It is known. Seems like it should be working in the next alpha build or you can check a nightly (bleeding) build

GUVWAF, 2024-06-15:
Note that you can only be connected to one client at a time. If the Android app is still connected, it will fight with the Web UI for the packets.