meshtastic/firmware #4119: [Bug]: Raspberry PI meshtasticd_2.3.13.a38a18d_amd64.deb doesnt recognize existing dependences

Plantucha, 2024-06-16:
### Category

Hardware Compatibility

### Hardware

Other

### Firmware Version

screen off

### Description

pi@APRS:~ $ sudo apt update
Hit:1 http://deb.debian.org/debian bookworm InRelease
Hit:2 http://deb.debian.org/debian-security bookworm-security InRelease
Hit:3 http://deb.debian.org/debian bookworm-updates InRelease
Hit:4 http://archive.raspberrypi.com/debian bookworm InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
pi@APRS:~ $ sudo apt dist-upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

pi@APRS:~ $ sudo apt install ./meshtasticd_2.3.13.a38a18d_amd64.deb
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'meshtasticd:amd64' instead of './meshtasticd_2.3.13.a38a18d_amd                 64.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 meshtasticd:amd64 : Depends: libyaml-cpp0.7:amd64 but it is not installable
                     Depends: libulfius2.7:amd64 but it is not installable
E: Unable to correct problems, you have held broken packages.

pi@APRS:~ $ sudo apt install libyaml-cpp0.7
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libyaml-cpp0.7 is already the newest version (0.7.0+dfsg-8+b1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
pi@APRS:~ $ sudo apt install libulfius2.7
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libulfius2.7 is already the newest version (2.7.13-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

madeofstown, 2024-06-16:
You tried to install the AMD64 .deb on an ARM platform.

closed 2024-06-16 by caveman99

roens, 2024-06-24:
I'm seeing a similar unmet dependency error, while attempting to install `meshtasticd_2.3.13.83f5ba0_arm64.deb` on an RPi 3B, running `Raspbian GNU/Linux 11 (bullseye)`.

I first installed dependencies mentioned on the [Linux Native Devices](https://meshtastic.org/docs/hardware/devices/linux-native-hardware/) page:

    sudo apt install libgpiod-dev libyaml-cpp-dev libbluetooth-dev openssl libssl-dev libulfius-dev liborcania-dev

I then downloaded `meshtasticd_2.3.13.83f5ba0_armhf.deb` from this repo's Releases page, then tried to install it:

```
sudo apt install ./meshtasticd_2.3.13.83f5ba0_armhf.deb 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'meshtasticd' instead of './meshtasticd_2.3.13.83f5ba0_armhf.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 meshtasticd : Depends: libyaml-cpp0.7 but it is not installable
E: Unable to correct problems, you have held broken packages.
```

It seems `libyaml-cpp0.7:arm64` is unavailable?:

```
apt-cache madison libyaml-cpp0.7
N: Unable to locate package libyaml-cpp0.7
```

But other versions are:

```
apt search libyaml-cpp
Sorting... Done
Full Text Search... Done
libyaml-cpp-dev/oldstable,now 0.6.3-9+rpi1 armhf [installed]
  YAML parser and emitter for C++ - development files

libyaml-cpp0.3-dev/oldstable 0.3.0-1.2 armhf
  YAML parser and emitter for C++ - development files (0.3 series)

libyaml-cpp0.3v5/oldstable 0.3.0-1.2 armhf
  YAML parser and emitter for C++ (0.3 series)

libyaml-cpp0.6/oldstable,now 0.6.3-9+rpi1 armhf [installed,automatic]
  YAML parser and emitter for C++
```

Everything is otherwise up-to-date; the following result in no package installs/upgrades/fixes/held-packages:

- `apt update && apt upgrade`
- `dpkg --force-all --configure -a`
- `apt --fix-broken install`
- `apt-get -f install`
- `dpkg -l | grep '^hi'`

And for the hell of it, I also tried installing `meshtasticd_2.3.13.83f5ba0_arm64.deb`, which resulted in one additional unmet dependency: `libulfius2.7:arm64 but it is not installable`

jp-bennett, 2024-06-24:
@roens at this point we're only providing .deb packages for Bookworm. For Bullseye you'd need to compile it yourself (or switch to Bookworm)

roens, 2024-06-24:
@jp-bennett Good to know, thanks for the quick reply!

roens, 2024-06-24:
Feel free to point me elsewhere or ignore… but as I'm trying to build/install _on a headless RPi_, I'm not using Visual Studio.

Where might one find CLI-focused build instructions for this repo? I'm guessing I'd install & use `pio`. And I can probably work it out, but I can probably work it out (likely by manually performing the contents of `.github/workflows/package_raspbian_armv7l.yml`).