meshtastic/firmware #4120: [Feature Request]: --request-environment

joneyes, 2024-06-16:
### Platform

other

### Description

I raised this in the python section first and was referred back to firmware:

It is possible to request the following via CLI for a remote node:
--request-position (returns position co-ords & alt)
--request-telemetry (returns current device metrics)

Is it possible to add --request-environmental to query environmental metric data, when enabled?

garthvh, 2024-06-16:
Should already be in the firmware, would need to send a environment telemetry packet to a node as a a dm and request a response

joneyes, 2024-06-16:
Thanks Garth,
any idea what the syntax is for sending an environment telemetry packet is? - i did look and couldn't see anything that jumped out... Hence the post.

GUVWAF, 2024-06-21:
Replies are not yet implemented in the firmware for Environmental Telemetry.

closed 2024-06-23 by thebentern

joneyes, 2024-07-18:
This is marked as completed.
How do you intiate the query and what is the syntax, as i can't find any supporting documentation?
I was looking for a way to do this to a remote node, so is it via a DM, or CLI --request- command?

GUVWAF, 2024-07-18:
It's implemented in the firmware. You can send a packet with the corresponding telemetry tag and set `want_response` to true, then it will respond. 

If you want to have it in the CLI, please open a feature request in that repository.

joneyes, 2024-07-18:
> It's implemented in the firmware. You can send a packet with the corresponding telemetry tag and set `want_response` to true, then it will respond. 
> 
> If you want to have it in the CLI, please open a feature request in that repository.

👍
Did that there before raising it here.

PR #4164 (2024-06-23): Implement replies for all telemetry types based on variant tag
This allows nodes to request all different telemetry types based on the variant tag in the payload (closes #4120). I had to decode the payload in `allocReply()` to check the tag. Also created `getTelemetry()` methods that return `true` if there is valid data, which is used in `sendTelemetry()` and `allocReply()`.

Tested with DeviceMetrics, and also checked that a NAK with `NO_RESPONSE` is sent when there's no data available. I couldn't easily test real data with Environment, AirQuality and Power, but logic is the same.

Also remov [... cut]