meshtastic/firmware #4132: [Feature Request]: Cancel canned message with main button

BRNSystems, 2024-06-17:
### Platform

NRF52, ESP32, RP2040, Linux Native, other

### Description

I would appreciate if pressing the main button(the one that cycles screens) would cancel the canned message sending, so when I hit the button on the rotary encoder accidentally, it won't send a message on it's own.

And could there be an option to turn on a confirmation dialog to make sure I don't send a message accidentally?

HarukiToreda, 2024-06-23:
I understand your concerns about accidental presses. While implementing a confirmation dialog or button press cancellation in the firmware could address this issue, it might also introduce delays that could be problematic in urgent situations.

It might be beneficial to consider a case design that makes accidental presses less likely. A well-designed case can help prevent unintentional button activations, maintaining the device's instant response capability, which is crucial for emergency scenarios.

BRNSystems, 2024-06-23:
And what about resetting the selector when the button 0 gets pressed? I find it annoying when I can't cancel a message selection.

closed 2024-06-23 by thebentern

Jvk1166z, 2025-08-10:
i really dont think that this issue should have been closed. i understand that having a confirmation dialogue might be against the design goal of being fast in an emergency, but i really dont see how having button 0 cancel the message would be in conflict with that. in fact, i think the way it's currently set up actually is detrimental to that design goal.

as it stands right now, if you open the canned messages screen, you can press button 0 to navigate away from that screen. the only problem with that is, when you do that, if you go to use the rotary encoder again, it wont actually put you back on the screen because it's still open at the front of the stack, requiring you to use the user button to navigate back to the canned message screen to pick and send your messages. depending on your device, this can take a few seconds, which in an emergency, could be a long time. a longer time for sure than having to go through a confirmation dialogue. if button 0 cancelled sending a message, this wouldn't be an issue.

additionally, the functionality for cancelling a message is already present in meshtastic, if you have a keyboard attached to your device, you can press the escape button to cancel sending a canned message, please educate me if i'm wrong about this, but based on the feature basically already being present, i don't see how it would be a big issue to simply allow button 0 to activate it when on the canned messages screen, or at least allowing some number of consecutive presses/length of button hold to do the same thing.