meshtastic/firmware #4144: [Bug]: QMC5883L Has the wrong address in the config

Africmod, 2024-06-19:
### Category

Hardware Compatibility

### Hardware

Not Applicable

### Firmware Version

2.3.12.24458a7

### Description

According to this datasheet: http://wiki.sunfounder.cc/images/7/72/QMC5883L-Datasheet-1.0.pdf
And also to my testing:
[image]

The basic address of the QMC5883L is 0x0D. The current configuration.h file suggest it should be 0x1E.

a57224812, 2024-08-24:
The basic address is 0x0D. It is suggested to include 0x1E, which one is correct

Africmod, 2024-08-24:
The address on the QMC5883L is 0X0D I think the configuration.h file should match that.

a57224812, 2024-08-29:
> QMC5883L上的地址是 0X0D我认为 configuration.h 文件应该匹配它。
Thank you for your answer, but I can't drive it now whether I use 0X0D or 0X1E. I don't know what the reason is

fifieldt, 2024-09-16:
@caveman99 - you're in git blame for the 0x1E address for the QMC5883L. Did you have one to test with, or should we go with what the docs say?

caveman99, 2024-09-16:
I have one but there's no support, I am just adding the sensor to detect. The actual implementation was done by somebody else it seems. I don't remember if I had it ever working.

a57224812, 2024-09-17:
I'm glad to see your answer. I'm currently trying to implement this feature

杰
***@***.***

closed 2024-09-17 by caveman99