meshtastic/firmware #4158: [Bug]: Battery info shows 0%

jranma, 2024-06-22:
### Category

Hardware Compatibility

### Hardware

T-Beam S3

### Firmware Version

2.3.12

### Description

Using t-beam S3 supreme, using using USB power supply. 
In the list of nodes, 0% 0.00V is shown.
instead of showing the logo that it is connected to the power socket.

fifieldt, 2024-07-03:
Can you try the ADC Multiplier Override calibration process? https://meshtastic.org/docs/configuration/radio/power/#calibration-process-attribution

jranma, 2024-07-08:
> Can you try the ADC Multiplier Override calibration process? https://meshtastic.org/docs/configuration/radio/power/#calibration-process-attribution

unfortunately, didnt help. playing with power settings doesnt help. 

does anyone have a clue about this? 
Just made a clean reflash, using 2.3.13. 
USB power not detected using factory settings.  
display: 
- UNK%
- UNK volts

jranma, 2024-07-08:
@thebentern could be related to incorrect INA219 Address ?

ayysasha, 2024-07-24:
Same issue here with 19003/v2.3.15.deb7c27

gitbisector, 2024-09-05:
My t-beam S3 supreme runs 2.4.1 and showing the plug icon (4.18V) on both the display and IOS app. Do you have a 18650 cell in the back too? Am using an Samsung unprotected 18650.

gitbisector, 2024-09-06:
Seems to be an IOS app thing. 
It shows battery 0% when on USB without battery pluggen in. When the battery is present it shows 'PWD' on the app.
Debug output from the T-Beam S3 supreme follows.

With the battery removed:
`DEBUG | 20:10:56 89 [Power] Battery: usbPower=1, isCharging=0, batMv=0, batPct=0`
With the battery present: 
DEBUG | 20:12:36 189 [Power] Battery: usbPower=1, isCharging=1, batMv=4184, batPct=14%

garthvh, 2024-09-06:
The device metrics packet is a percentage from 0-101 with 101 meaning powered.

closed 2024-09-06 by garthvh

gitbisector, 2024-09-06:
While on USB power, without a battery present, the wrong status is getting reported. Should probably fix it then to send '101' in this state as the device is able to recognize the difference.

jranma, 2024-09-06:
Indeed issue happens only when no battery is attached.

PR #4646 (2024-09-07): Report PWD when no battery present.
Fixes #4158.

Report back MAGIC_USB_BATTERY_LEVEL (i.e. 'PWD')  when there is no battery present.