meshtastic/firmware #4166: [Bug]: store and forward chronology

jranma, 2024-06-23:
### Category

Other

### Hardware

T-Beam S3

### Firmware Version

2.3.12

### Description

when one retrieves store and forward messages using "SF", all messages will arrive at the same time. Above all, they will all be displayed with the time they were retrieved, not the time they were received on the relay. It is therefore difficult, if not impossible, to understand the thread of the conversation because the chronologies are wrong.

garthvh, 2024-06-28:
Where are you seeing the messages with the same time?

jranma, 2024-06-30:
each time I request messages with a "SF" command, I get all the messages with the same time, which is the time at wich I requested them. 
Using Android APP.

GUVWAF, 2024-06-30:
Yes, that sounds correct. Currently the timestamp when the packet was actually received is not stored and send with the packet upon request.

jranma, 2024-06-30:
OK, not sure the messages come in the right order. 
Last time I received many messages and as the timestamp are all the same, I had a hard time understanding the logic of the conversation. 
Having the message sorted by their "real" timestamp would be a nice emprovement.

GUVWAF, 2024-07-07:
I think we can make this work. Currently we do store a relative time in the StoreForward history, but we can change it to an absolute time if we have one, and then not updating the `rx_time` for StoreForward packets when requesting them.