meshtastic/firmware #4175: [Bug]: Device becomes unreachable Heltec & LILYGO

TheDomanNow, 2024-06-25:
### Category

BLE, WiFi

### Hardware

Heltec V3, Other

### Firmware Version

2.3.12

### Description

Hi,

currently I have 2 "base" nodes in good position & with constant power, ROUTER_CLIENT mode. They are connected via WiFi to internet and have MQTT enabled. After like 2-3 days they become unresponsive via IP address (HTTP or Android app). This is not an network (WiFi) issue, also not a hardware issue (1x Heltec V3.1 & 1x LILYGO T3S3).
In this unreachable state it is possible to send a message via other node that is currently connected, so the radio is working. After power cycling the device all is back to normal, but all messages on router are lost...

I'm guessing this is related with current faulty implementation of sleep (?). Is there any workaround of this??? Is it possible to completely turn off all power saving "features"?

I see 2 solutions (probably both should be added):
1. Adding a toggle, like "Turn off all power saving things", that really turns off all power saving "features".
2. Adding auto power cycle feature, that after X time or at desired hour (internet needed) restarts the device.

BR, Doman.

madeofstown, 2024-06-25:
Is the MQTT channel very active? About how many nodes are you seeing on your node list? Busy MQTT channels are know to overload these devices.

HarukiToreda, 2024-06-26:
Power save mode settings are not on by default, if you did not set them, the device will not go to lite sleep as you're suspecting. The only thing that will turn off is the screen and it should not be that either. I believe mqtt is flooding the node with too many nodes. To work around this, I suggest using admin to be able to control your node from other nodes via the app.

TheDomanNow, 2024-06-26:
> Is the MQTT channel very active? About how many nodes are you seeing on your node list? Busy MQTT channels are know to overload these devices.

No it isn't. NodeDB is almost full or full, but there are like 30 nodes active at a time, sending maybe 1x msg/hour. It's not that.

> Power save mode settings are not on by default, if you did not set them, the device will not go to lite sleep as you're suspecting. The only thing that will turn off is the screen and it should not [...]

So, you're saying that settings like:
"Wait for Bluetooth duration"
"Light sleep duration"
"Minimum wake time"
are NOT active, unless "Enable power saving mode" is turned ON? Why aren't they greyed out then?

Like said above, MQTT doesn't have much traffic and nodes, it's not flooded. It's not that.

I still thing it's related to some power saving "feature". Or... any other ideas?