meshtastic/firmware #4177: [Bug]:

Oliver0804, 2024-06-26:
### Category

Other

### Hardware

T-Echo

### Firmware Version

2.3.12 and 2.3.13

### Description

I am experiencing unexpected crashes with the Techo firmware in versions 2.3.12 and 2.3.13. These crashes cause the following issues:

The physical buttons become unresponsive.
The DFU functionality fails to start.
Bluetooth connectivity is lost.
Current Workaround:

The only way to recover from this state is by using the reset button.

Additional Information:

After resetting, I have reverted to version 2.3.11, which is stable and currently in use without these issues.

Steps to Reproduce:

Update Techo firmware to version 2.3.12 or 2.3.13.
Observe the system for unexpected crashes and unresponsiveness.
Expected Behavior:

The Techo device should function without crashing, and all features (buttons, DFU, Bluetooth) should remain operational.

Actual Behavior:

The device crashes unpredictably, leading to the described issues.

Thank you.

todd-herbert, 2024-06-26:
How frequently are you seeing this? I might have seen something similar once, with several days of uptime, but I've seen this happen prior to 2.3.12 also.

If you haven't tried it already, I'd definitely suggest [flashing the erase .uf2 file](https://meshtastic.org/docs/getting-started/flashing-firmware/nrf52/nrf52-erase/) and making sure you've got the [Meshtastic bootloader](https://meshtastic.org/docs/getting-started/flashing-firmware/nrf52/update-techo-bootloader/).

Oliver0804, 2024-06-26:
Versions 2.3.12 and 2.3.13 will occur approximately within 24 hours.

Additionally, you mentioned erasing the entire system before updating to version 2.3.12? Currently, I have only been updating without erasing user settings.

If this was how you resolved the issue before, I will give it a try. Thank you.

todd-herbert, 2024-06-26:
> Currently, I have only been updating without erasing user settings.

Usually this seems okay, but if I get a strange error, erasing is the first thing I try. Quite often it fixes it! Hopefully it helps here too. Do try updating your bootloader too if you haven't before.

Oliver0804, 2024-06-26:
> > Currently, I have only been updating without erasing user settings.
> 
> Usually this seems okay, but if I get a strange error, erasing is the first thing I try. Quite often it fixes it! Hopefully it helps here too. Do try updating your bootloader too if you haven't before.
> 
> Also, should probably ask for as much info as possible now! Is there anything special about the way you're using your device? Are you down-linking MQTT?

I will first try updating the bootloader. Now I am updating to version 2.3.13 and will observe for a few days.

Additionally, I have two channels with both MQTT up/down enabled. Will this have any effect? The T-ECHO is not necessarily always with the phone. Could this be caused by MQTT?

[image]

todd-herbert, 2024-06-26:
I don't think there is a known bug with MQTT or anything, but I did see https://github.com/meshtastic/firmware/discussions/4179 appear yesterday, so it made me wonder!

micheljung, 2024-06-27:
My T-Echo didn't respond after installing 2.3.13, only the red light flashed. Tried 2.3.12 and it worked. Tried 2.3.13 again and got the same issue again. In the logs I find:

```
INFO | ??:??:?? 8 Set radio: final power level=22
INFO | ??:??:?? 28 SX126x init result -2
WARN | ??:??:?? 28 Failed to find SX1262 radio
INFO | ??:??:?? 28 Not using Ethernet
ERROR | ??:??:?? 28 NOTE! Recording critical error 3 at src/main.cpp:976
```

todd-herbert, 2024-06-27:
@micheljung Now that *is* interesting. The RadioLib version *did* change with 2.3.12, but I'm not seeing anything like this going on with my T-Echo, so I'm not sure what to make of it. Hopefully one of the radio experts drops by.

Have you also had a go with this?
> If you haven't tried it already, I'd definitely suggest [flashing the erase .uf2 file](https://meshtastic.org/docs/getting-started/flashing-firmware/nrf52/nrf52-erase/) and making sure you've got the [Meshtastic bootloader](https://meshtastic.org/docs/getting-started/flashing-firmware/nrf52/update-techo-bootloader/). If you've never flashed this bootloader, it's worth trying, even if the version is reported as the same, as it seems to be a slightly different build than the one the devices ship with.

Oliver0804, 2024-07-07:
The bootloader has been updated, and there have been no crashes for about a week now, so I'm closing this issue. Additionally, I recommend notifying users to update their bootloaders if there are any modifications to it.

closed 2024-07-07 by Oliver0804

todd-herbert, 2024-07-07:
I don't think it's been updated since February, but it does seem to be related to these weird issues, at least sometimes. Glad to hear it's looking stable, for now at least!