meshtastic/firmware #4180: Warnings generated during compilation

thebentern, 2024-06-26:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

Current master

### Description

Warnings generated during compilation
src/graphics/Screen.h:524:13: warning: 'void drawCompassNorth(OLEDDisplay*, int16_t, int16_t, float)' defined but not used [-Wunused-function]
 static void drawCompassNorth(OLEDDisplay *display, int16_t compassX, int16_t compassY, float myHeading)

jp-bennett, 2024-06-26:
@thebentern What target throws this warning? And the broader question, how do we want to handle functions that *could* be useful for multiple targets, but isn't used for all of them?

thebentern, 2024-06-26:
Multiple targets. Heltec V3 is a good example. 
We can guardclause it if need-be.

jp-bennett, 2024-06-26:
It seems like it's complaining because it's a static function, that's not part of the Screen class. Maybe the right answer here is to make it a normal class function instead.

closed 2024-07-24 by thebentern