meshtastic/firmware #4181: [Bug]: Occasionally restarting after sending message

mattdrum, 2024-06-26:
### Category

Other

### Hardware

T-Deck

### Firmware Version

2.3.13

### Description

Sometimes after sending a message when the screen is saying Delivered to ---, the Tdeck restarts.

memo-567, 2024-07-21:
Same here on two T-Deck devices  ¯\\\_(ツ)_/¯

fifieldt, 2024-09-16:
Thanks for the report. Are you able to provide any serial logs?

http://flasher.meshtastic.org --> Open Serial Monitor

is one way to get them

fifieldt, 2024-09-16:
Thanks for the rapid reply and sorry to hear about the response to date :(

GUVWAF, 2024-09-16:
> After these errors were ignored for a very long time

I'm sorry, but I've also asked you on Discord for serial logs and your configuration, because I don't have a T-Deck and developers that have one are mostly working on the new UI. You didn't provide any, so I couldn't help you further.

The fact that only two users have reported this serious issue so far suggests it only happens with a specific configuration/scenario, or it has been resolved already.

closed 2024-09-16 by thebentern

joyel24, 2024-12-16:
I suspect a link with this issue #4275...