meshtastic/firmware #4182: [Bug]: very long "long name"

jranma, 2024-06-26:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

2.3.13

### Description

[image]

Some user added many "/n" to his "long name", which causes display problems on every app. 
Shouldn't these special control characters be filtered?

garthvh, 2024-06-26:
[image]

I would have to allow them to work as line breaks on iOS

Cisien, 2024-06-29:
The issue you're seeing has nothing to do with `/n` (which is just two normal characters and not a special escape sequence)

I tried reproducing by setting the name using several variations via the python cli with no impact on how the nodes table rendered.

[image]

[image]

[image]

garthvh, 2024-06-29:
They are newlines

garthvh, 2024-06-29:
You can easily replicate this with 
meshtastic --set-owner '''tbeamer Line1
Line2
Line3'''

Cisien, 2024-06-29:
`\n` is a new line,  `/n` is two regular ascii characters
and yes, i can replicate this on my nodes (images in the discord channel)

garthvh, 2024-06-29:
@thebentern I think we need to replace \n with a space

Cisien, 2024-06-29:
may want to do the same with other characters. `\t` for example does some "interesting" things
[image]
in the IOS app it just looks like extra spaces

`--set-owner "Cisien $(echo -e '\t\t')Rak"`

closed 2024-07-13 by garthvh