meshtastic/firmware #4197: [Feature Request]: Set Maximum Nodes for Device or NodeDB

lagunacomputer, 2024-06-28:
### Platform

ESP32

### Description

[Feature Request]: Set Maximum Nodes for Device or NodeDB

Just throwing this idea out there.  It would be great to be able to set a 'Maximum Nodes Allowed' setting for the discovered node list or NodeDB, either on the device or in the app.

I know that in recent firmwares that NodeDB deletes the oldest entry when full, however I am finding my 'just seems to work better' device works better when the NodeDB isn't full (possibly device out of free space or ram?). It is not uncommon for my device to have 225+ modes in its list, even with MQTT off.  

An easy fix could be to just be able to manually limit the Node list or NodeDB size.  My device is a TBeam clone custom hardware ESP32.  Just throwing this idea out there, thanks!

jp-bennett, 2024-06-28:
Check out the MAX_NUM_NODES defines

lagunacomputer, 2024-06-28:
I will give it a shot in my variant.h , Thx!

closed 2024-06-28 by thebentern

roha-github, 2024-06-29:
I would rather wish that you could have 1000 nodes and more in the mesh.

Perhaps the problem lies more in the broadcast / flooding of static node info & co. In my opinion, new nodes could retrieve the missing nodeinfo from their neighbors if necessary, instead of distributing it regularly across all nodes in the mesh. The mesh is more busy sending metadata than real messages from people.