meshtastic/firmware #4206: [Bug]: USB Serial does not recover when waking from light sleep on ESP32-S3 devices using native ESP32 serial over usb

Cisien, 2024-06-30:
### Category

Serial

### Hardware

T-Deck, Other

### Firmware Version

2.3.11+

### Description

When the esp32 enters light sleep mode the usb serial connection does not always recover. This often leads to the USB connection getting stuck in a bad state until the device is rebooted.

Validated the behavior by setting the `device.role` to `ROUTER` and `display.screen_on_secs` to `15`
Can reproduce on Station G2 and T-Deck
Reproduces on 2.3.11 and 2.3.14

Changing the devices role from ROUTER to CLIENT mitigates the issue by not attempting to enter light sleep

### Relevant log output

```Shell
DEBUG | ??:??:?? 55 [RadioIf] AirTime - Packet transmitted : 641ms
INFO  | ??:??:?? 55 [PowerFSM] setup LORA_DIO1 (GPIO45) with wakeup by gpio interrupt
NFO  | ??:??:?? 56 [PowerFSM] Exit light sleep gpio: btn=0
INFO  | ??:??:?? 56 [PowerFSM] Exit state: LS
```
(not ehte missing `I` in line 3 above)

A few moments later, serial dropped long enough to timeout the meshtastic cli (`--debug --noproto`)
```
DEBUG | 01:37:32 85 [Power] Battery: usbPower=1, isCharging=1, batMv=4700, batPct=100
INFO  | 01:37:38 92 [PowerFSM] lsEnter begin, ls_secs=86400
WARNING file:stream_interface.py __reader line:187 Meshtastic serial port disconnected, disconnecting... device reports readiness to read but returned no data (device disconnected or multiple access on port?)
DEBUG file:stream_interface.py __reader line:202 reader is exiting
DEBUG file:stream_interface.py _disconnected line:80 Closing our port
```
dmesg:
```
[441723.589803] usb 1-5.2: USB disconnect, device number 97
[441753.764416] usb 1-5.2: new full-speed USB device number 98 using xhci_hcd
[441753.864861] usb 1-5.2: device descriptor read/all, error -32
[441753.944412] usb 1-5.2: new full-speed USB device number 99 using xhci_hcd
[441754.028430] usb 1-5.2: device descriptor read/64, error -32
[441754.216428] usb 1-5.2: device descriptor read/64, error -32
[441754.324839] usb 1-5-port2: attempt power cycle
[441754.928404] usb 1-5.2: new full-speed USB device number 100 using xhci_hcd
[441754.928498] usb 1-5.2: Device not responding to setup address.
[441755.136473] usb 1-5.2: Device not responding to setup address.
[441755.344393] usb 1-5.2: device not accepting address 100, error -71
[441755.344810] usb 1-5.2: WARN: invalid context state for evaluate context command.
[441755.424393] usb 1-5.2: new full-speed USB device number 101 using xhci_hcd
[441755.424474] usb 1-5.2: Device not responding to setup address.
[441755.632479] usb 1-5.2: Device not responding to setup address.
[441755.840386] usb 1-5.2: device not accepting address 101, error -71
[441755.840808] usb 1-5.2: WARN: invalid context state for evaluate context command.
[441755.840856] usb 1-5-port2: unable to enumerate USB device
[441791.907851] usb 1-5.2: new full-speed USB device number 102 using xhci_hcd
[441791.991919] usb 1-5.2: Device not responding to setup address.
[441792.199893] usb 1-5.2: Device not responding to setup address.
[441792.407862] usb 1-5.2: device not accepting address 102, error -71
[441792.408252] usb 1-5.2: WARN: invalid context state for evaluate context command.
[441792.695844] usb 1-5.2: new full-speed USB device number 103 using xhci_hcd
[441792.817850] usb 1-5.2: New USB device found, idVendor=303a, idProduct=1001, bcdDevice= 1.00
[441792.817855] usb 1-5.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[441792.817859] usb 1-5.2: Product: BQ Station G2
[441792.817861] usb 1-5.2: Manufacturer: Espressif Systems
[441792.817863] usb 1-5.2: SerialNumber: ECDA3B46B8F8
[441792.877908] cdc_acm 1-5.2:1.0: ttyACM0: USB ACM device
```

Discussion in discord: https://discord.com/channels/867578229534359593/1039674919346249728/1256780634492506193
```

Cisien, 2024-06-30:
Speaking further with @thebentern in discord, this is likely by design. The connection _does_ come back on activity as expected when using a real serial application instead of the CLI.

The main issue I guess i have with this behavior is that initiating a connection via serial or the CLI does not wake the device, and an active connection doesn't keep it from sleeping.

closed 2024-10-07 by thebentern

DrDoof, 2026-08-20:
For anyone landing here from a search: this still reproduces on 2.7.26, and I think I found the cause. Opened #11556 with a fix.

Short version: role ROUTER sets `isPowerSavingMode = config.power.is_power_saving || isRouter` in `isPowered()`, which forces power saving even when the board is plugged into USB. The light sleep transitions then get added, and on ESP32-S3 with `ARDUINO_USB_MODE=0` light sleep powers down the native USB-OTG PHY, so the host can never re-enumerate the device. That is where the `error -32` loop comes from.

That also explains why switching to CLIENT helped, as noted in the original report. It does not enter light sleep, so the PHY stays up.

The fix skips light sleep only when the board uses native USB-CDC and `getHasUSB()` confirms it is running on USB power, so battery and solar routers keep the existing behaviour. Verified on a T-Beam S3 Core: 20 minutes in role ROUTER, zero USB errors, where before it died within about 30 seconds.