meshtastic/firmware #4207: [Bug]: Neighbor info packets are not uplinked to MQTT when the neighbor info module is disabled

armooo, 2024-06-30:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

2.3.14.64531fa

### Description

1) Enable the MQTT module.
2) Enable MQTT uplink for a channel.
3) Have another node send a neighbor info packet.
4) See the packet is not published to MQTT.
5) Enable the neighbor info module on the node with MQTT uplink enabled.
6) Observe the packets are now published.

I suspect this is due to Router.cpp ignoring NEIGHBORINFO_APP when the module is disabled. See: https://github.com/meshtastic/firmware/blob/47a94d7a076e2cd9ba0e3260ca51a74a7c101f01/src/mesh/Router.cpp#L463-L469

garthvh, 2024-06-30:
Yes neighbor info is opt in, if it is disabled it is not routed over Lora or mqtt

closed 2024-06-30 by garthvh

armooo, 2024-06-30:
Seem unfortunate I need to spam packets on RF to be able to upload the neighbor info to MQTT.

GUVWAF, 2024-06-30:
I think creating an "MQTT-only" setting for the NeighborInfo module might make sense. It seems most people currently use it to see them on a map, for which you indeed don't need to send it over LoRa.

prokrypt, 2024-06-30:
I think @garthvh is misunderstanding the problem here: nodes with neighborinfo module disabled will not upload any received neighborinfo packets to mqtt. I think this is indeed a bug.

GUVWAF, 2024-06-30:
The reason for this is because NeighborInfo packets are deliberately not rebroadcasted on the mesh if you don't have it enabled. I think that's a good measure, but the consequence is that it will then also not send it via MQTT.

armooo, 2024-06-30:
The one problem I see is that `last_sent_by_id` will inconsistent between nodes with the neighbor info module enabled and disabled. I have been running with the MQTT onSend call outside of the `!skipHandle` check on mode MQTT uploading node now.

My use case is to uplink the packets from a few nearby routers that are running with 0 or 1 hops.

GUVWAF, 2024-06-30:
Yeah, but `last_sent_by_id` is not needed anymore since we added `hopStart`. Neighbors without NeighborInfo module enabled can be discovered when `hopStart == hopLimit`.

armooo, 2024-06-30:
I think you still need `last_sent_by_id` if a node is set to 0 hops because you can't tell the proto zero value of an old firmware from a node set to 0 hops.

GUVWAF, 2024-06-30:
True, but to be honest that's not a common practice.

garthvh, 2024-07-01:
> I think @garthvh is misunderstanding the problem here: mqtt nodes with neighborinfo module disabled will not upload any received neighborinfo packets to mqtt. I think this is indeed a bug.

I get it, these hybrid options keep burning us, I think of it is disabled it should be disabled, not half disabled.

VigibotDev, 2024-08-05:
I made a real-time graph (public on https://www.serveurperso.com for our Parisian mesh) and GUVWAF says a very interesting tip, if I understand correctly, we should be able to generate the graph of an entire mesh without using the "Neighbor Info" module using all the frames with hopStart == hopLimit ?

armooo, 2024-08-16:
Ya, I already do server side neighbor info as green lines [here](https://meshview.armooo.net/graph/network?hours=2) but you can't do that with the public MQTT server anymore because the data is lies.

VigibotDev, 2024-08-16:
And it's like ROUTER don't refresh neighbor info on lasts versions 😱😱😱😱😱

GUVWAF, 2024-08-16:
> And it's like ROUTER don't refresh neighbor info on lasts versions 😱😱😱😱😱

By default the broadcast interval of `ROUTER`s is set to 12 hours.

VigibotDev, 2024-08-16:
> > And it's like ROUTER don't refresh neighbor info on lasts versions 😱😱😱😱😱
> 
> By default the broadcast interval of `ROUTER`s is set to 12 hours.

Thanks for this information, because we updated + reset a ROUTER node less than 12 hours.
Also the other ROUTER node I upgraded without reset, continue to send neighbor every 1 hours....
For supervision 1 hour is better and for tiny autonomous mountain node 12h consume less and it's better...
Meshtastic need more user choice to the ROUTER owner because here we made high power 19V solar in urban area tower roof don't have power problems...