meshtastic/firmware #4212: [Bug]: MQTT messages not going to private server

valzzu, 2024-06-30:
### Category

Other

### Hardware

Heltec V3

### Firmware Version

2.3.13.83f5ba0

### Description

My base node is connected to the public server and i have my heltec connected to my private mqtt server.

I found out that mqtt messages coming from my base are not sent to my private server.
I made a nodejs script so i could send data logged to discord but then i found it out.
Using json output.

Talie5in, 2024-07-01:
This is what MQTT Bridging is for, the idea would be Node <-> Private MQTT Broker <-> Bridge <-> Public MQTT Broker
Ideally, not allowing MQTT tagged packets to then be pushed to MQTT (eg, MQTT <->Node 1 <-> Node2 <-> MQTT) again would be to stop potential MQTT data loops.

garthvh, 2024-07-03:
This is intended as it reduced mqtt traffic by 85% and made it actually work.

closed 2024-07-03 by garthvh