meshtastic/firmware #4223: [Bug]: enabling GPS crashes wio-sdk-wm1110

fifieldt, 2024-07-02:
### Category

Hardware Compatibility

### Hardware

Other

### Firmware Version

2.3.15.84088f39

### Description

Set GPS to ENABLED in settings on wio-sdk-wm1110.

Log will halt at:

"Probing for GPS at 9600"

and no more logging can be seen.

### Relevant log output

```Shell
"Probing for GPS at 9600"
```

Dylanliacc, 2024-07-04:
Actually, it's not a bug. Our firmware does not support the nRF52's internal GPS module. We need to use an external Grove-GPS (Air530) to make sure the GPS function of WM1110 for Meshtastic works correctly.

Dylanliacc, 2024-07-04:
You can find more details here: [Grove-GPS Air530](https://wiki.seeedstudio.com/Grove-GPS-Air530/). I'll also provide pictures of the device connection, app configuration, and working logs.  
[image]
[image]
[image]

fifieldt, 2024-07-15:
@garthvh @Dylanliacc this is still an issue. Even with a Grove-GPS Air530 connected to the wio-sdk-wm1110, enabling GPS crashes the device. (NB: the wio-tracker-wm1110 pictured above works fine)

[meshtastic-log-2024-07-15T03-11-27.876Z.log](https://github.com/user-attachments/files/16229714/meshtastic-log-2024-07-15T03-11-27.876Z.log)

thebentern, 2024-07-15:
We need to check the serial definitions. @fifieldt do you know off-hand if the grove UART connector might pin with the UART connected to the CH340? Otherwise we probably need to declare a secondary Serial for the GPS to scan instead

fifieldt, 2024-07-15:
Taking a look...

fifieldt, 2024-07-15:
My working theory:

#define GPS_RX_PIN PIN_SERIAL2_RX
#define GPS_TX_PIN PIN_SERIAL2_TX
#define GPS_BAUDRATE 115200

based on:

[image]

fifieldt, 2024-07-15:
My attempt at setup code for the Air350:
```
+        } else if (gnssModel == GNSS_MODEL_AIROHA) {
+            // Set Normal Mode
+            _serial_gps->write("$PGKC105,0*37\r\n");
+            delay(250);
+            // Disable NMEA while we're configuring
+            _serial_gps->write("$PGKC242,0,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0*37\r\n");
+            // Set NMEAInterval to 1000ms
+            _serial_gps->write("$PGKC101,1000\r\n");
+            // Initialize the AIROHA Chip, use GPS + GLONASS + BEIDOU + GALILEO
+            _serial_gps->write("$PGKC115,1,1,1,1,0*2B\r\n");
+            delay(250);
+            // Enable NMEA, only ask for RMC and GGA
+            _serial_gps->write("$PGKC242,0,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0*37\r\n");
+            delay(250);
```

fifieldt, 2024-08-14:
The recent changes to the probe code have fixed this I believe

closed 2024-08-14 by fifieldt