meshtastic/firmware #4227: [Feature Request]: Networking and API in ROUTER mode

petrkr, 2024-07-03:
### Platform

NRF52, ESP32, RP2040, Linux Native, other

### Description

In order to disabling ROUTER_CLIENT mode (https://github.com/meshtastic/firmware/issues/4185) as side effect of this is huge breaking change, which will affect lot of nodes. That is networking and API is disabled in this mode. That will affect gathering statistic data from nodes and also OTA updating.

This feature requests this:
 - Allow optional networking in ROUTER mode (WiFi, Ethernet, ...)
 - Provide statistics data (nodes in db, metrics like routed messages, number of nodes, etc..)
   This can be done by:
    - API (just enable few api requests from as it is now)
    - SNMP
    - JSON on WEB
 - Allow forwarding received packets to MQTT ( compared to client nodes, this infrastracture nodes mostly run 24/7 and has better coverage to gather data)
 
Without this lot of services will have problem
 - meshmap.net will stop working, because it gather data from MQTT, where now Router_client nodes sending those data
 - monitoring of nodes will stop working
 - remote administration of nodes will stop working
 - remote updates will not be possible
 - monitoring of mesh network, which rely on data will stop work
 

This requests is NOT to have "client" mode in "router" node, but to have possibility to monitor and admin remote router nodes on high leveled places like observatories, cell towers and other hard-accessible places (mostly that where you need allowance from ISP or you need climb equipment)

pinkolin, 2024-07-03:
From my point of view, this is blocker feature for another stable release.

GUVWAF, 2024-07-03:
This is not a breaking change. `ROUTER_CLIENT` roles will be automatically assigned to `CLIENT`, which will have the networking and API capabilities. 
If you still want only these nodes to do the routing work, consider changing the other `CLIENT`s in the area to `CLIENT_MUTE`. This will also lower channel utilization and thus likely have better performance in general.

closed 2024-07-03 by garthvh

pinkolin, 2024-07-03:
If you will continue in this manner and not listen to us, we will must fork official release. I fully understand, OSS, you are not paid, but this is not how this things works. Czech community is one of  the biggest in the Europe and we have enough sources and resources to go own way. 

Reopen ticket. Last warning.

GUVWAF, 2024-07-03:
Forking is not necessary. A combination of `CLIENT`s and `CLIENT_MUTE` will work better than loads of `ROUTER_CLIENT`s and `CLIENT`s. Please first try this yourself.

pinkolin, 2024-07-03:
We understand, why is ROUTER_CLIENT bad for network. We still need ip network + API support in ROUTER mode. Where is problem? We have practical problem and you have feelings. This is problem.

thebentern, 2024-07-03:
> We understand, why is ROUTER_CLIENT bad for network. We still need ip network + API support in ROUTER mode. Where is problem? We have practical problem and you have feelings. This is problem.

petrkr, 2024-07-03:
> > We understand, why is ROUTER_CLIENT bad for network. We still need ip network + API support in ROUTER mode. Where is problem? We have practical problem and you have feelings. This is problem.
> 
> I don't believe ROUTER mode necessarily prevents IP / API modes from working. Have you tried maxing out `power.min_wake_secs`, `power.wait_bluetooth_secs` and `display.screen_on_secs`? Most things in Meshtastic can be granularly controlled if you do the research.

I tested it, it will connect for few seconds and disconnects, same for serial.

And no, router does not have remote administration. I can not remotely admin and control 9000km far away node on cell tower.

thebentern, 2024-07-03:
> I tested it, it will connect for few seconds and disconnects, same for serial.

This is not the case. Run `meshtastic --set display.screen_on_secs 4294967295 --set power.wait_bluetooth_secs 4294967295 --set power.min_wake_secs 4294967295`. I just tested this locally on a Heltec V3 esp32 to confirm. I have a Router node staying connected to wifi and python API connection to the device running steady over the network. The API service runs fine as does the wifi stack on Router. If you simply configure your nodes correctly with these intervals, you won't have any of the issues you're concerned about.

Current Python API has a bug that affects Router mode because it does not keep alive serial because it uses ls_secs to determine heartbeat. You can solve by running `meshtastic --set power.ls_secs 300`

OK9UWU, 2024-07-03:
> Forking is not necessary. A combination of `CLIENT`s and `CLIENT_MUTE` will work better than loads of `ROUTER_CLIENT`s and `CLIENT`s. Please first try this yourself.

My pardon, but what exactly does 'CLIENT_MUTE' do? The documentation on it is very vague, is it listen only client or what?
Because what you are recommending is to switch from 'ROUTER_CLIENT' where ROUTER part is relaying messages and CLIENT part is making the telemetry available to the internet, to just a CLIENT that doesn't do anything but listen to the traffic with no relaying (from what i understood), which is useless.

ROUTER mode itself by default enables power saving which disables BLE/WIFI etc. which can be disabled manually, but again, documentation wasn't 100% clear on this.

petrkr, 2024-07-03:
> > I tested it, it will connect for few seconds and disconnects, same for serial.
> 
> This is not the case. Run `meshtastic --set display.screen_on_secs 4294967295 --set power.wait_bluetooth_secs 4294967295 --set power.min_wake_secs 4294967295`. I just tested this locally on a Heltec V3 esp32 to confirm. I have a Router node staying connected to wifi and python API connection to the device running steady over the network. The API service runs fine as does the wifi stack on Router. If you simply configure your nodes correctly with these intervals, you won't have any of the issues you're concerned about.
> 
> Current Python API has a bug that affects Router mode because it does not keep alive serial API because it uses ls_secs to determine heartbeat. You can solve by running `meshtastic --set power.ls_secs 300`

That is workaround on which we can not rely. They may introduce 'max value' right in next release as they removed RC mode. So for now maybe it's good workaround, but definively not solution.

MultiTricker, 2024-07-03:
I have a solo device which is indeed in advantageous position so `ROUTER` is a way to go, but I still need API to remotely control it, update it and monitor it.

In footnotes of documentation here:
https://meshtastic.org/docs/configuration/radio/device/

Is exact recommendation: "The Router role enables Power Saving by default. Consider ROUTER_CLIENT if BLE/WiFi/Serial are still needed."

And that's basically the main issue which I believe is still not addresed and that's why this Issue was created (and therefore should not be closed).

GUVWAF, 2024-07-03:
Then for this specific case (if the other nodes can't use `CLIENT_MUTE`) it makes sense to set the `wait_bluetooth_secs` and `min_wake_secs` high as suggested above. It's not a workaround, it will be the recommended way for these specific cases from now on. 
You don't need to be afraid this will be removed as well, and if that happens, then it's valid to raise an issue.

MultiTricker, 2024-07-03:
That would be great! It's not straigtforward, but I can surely live with that. Will test that when new firmware is out to be sure it works as you describedon that exact version (and so on). Thank you!

MultiTricker, 2024-07-03:
> The docs will be updated. The router_client deprecation isn’t in a release yet so that’s why the docs haven’t been updated yet.

Got it, thanks.