meshtastic/firmware #4236: MQTT code is taking up sizable (12K) flash space even on non TCP-IP (nrf52) boards.

geeksville, 2024-07-05:
### Category

Other

### Hardware

Rak4631

### Firmware Version

2.3.15.45bb9531

### Description

Seen on rak4631 target today with platform IO inspect tool.  Something is probably wrong with our build flags, because on a NRF52 the MQTT code should not be consuming space.  Which is a big deal because flash is almost full.

If no one else looks into this first I'll investigate in a couple of weeks probably.

[image]

geeksville, 2024-07-05:
oh nevermind - the rak4631 can have ethernet.

closed 2024-07-05 by geeksville

ianmcorvidae, 2024-07-05:
That ethernet is currently somewhat broken, at least the TCP client API part. I guess we don't want an explosion of different targets, but I could definitely see the argument that a separate variant for RAKs with ethernet is a good idea, especially if we're looking to save space for other stuff. (And the other bug would be good to fix too probably, heh)