meshtastic/firmware #4240: [Bug]: MQTT on "Portduino / Linux" Connection Cycles

blu006, 2024-07-06:
### Category

Other

### Hardware

DIY

### Firmware Version

2.3.13.83f5ba0

### Description

The MQTT connection configuration on Portduino only works through the Android app, and it continuously connection cycles on the public MQTT server at `mqtt.meshtastic.org`. 

My device is an orangepione.

`armbianmonitor -u` reports the following:
```

### lsusb:

Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=musb-hdrc/1p, 480M
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=ohci-platform/1p, 12M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=ohci-platform/1p, 12M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/1p, 480M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/1p, 480M

### Group membership of brandon : brandon tty disk dialout sudo audio video plugdev games users systemd-journal input render netdev

### Userland generated with Armbian Build Framework:

PRETTY_NAME="Armbian 24.5.1 bookworm"
ARMBIAN_PRETTY_NAME="Armbian 24.5.1 bookworm"

### Installed packages:

ii  armbian-bsp-cli-orangepione-current 24.5.1                         armhf        Armbian CLI BSP for board 'orangepione' branch 'current'
ii  armbian-firmware                    24.5.1                         all          Armbian - Linux firmware
ii  linux-base                          4.9                            all          Linux image base package
ii  linux-dtb-current-sunxi             24.5.1                         armhf        Armbian Linux current DTBs in /boot/dtb-6.6.31-current-sunxi
ii  linux-image-current-sunxi           24.5.1                         armhf        Armbian Linux current kernel image 6.6.31-current-sunxi
ii  linux-libc-dev:armhf                6.1.94-1                       armhf        Linux support headers for userspace development
ii  linux-u-boot-orangepione-current    24.5.1                         armhf        Das U-Boot for orangepione

### Loaded modules:

Module                  Size  Used by
pps_ldisc              16384  2
bluetooth             528384  10
ecdh_generic           12288  1 bluetooth
ecc                    24576  1 ecdh_generic
rfkill                 20480  2 bluetooth
snd_soc_hdmi_codec     16384  1
lz4hc                  12288  0
sunxi_cedrus           40960  0
v4l2_mem2mem           20480  1 sunxi_cedrus
sun9i_hdmi_audio       12288  0
sun4i_gpadc_iio        12288  0
lima                   49152  0
lz4                    12288  0
videobuf2_dma_contig    16384  1 sunxi_cedrus
industrialio           57344  1 sun4i_gpadc_iio
[... 570 lines cut ...]
Jul 06 02:51:07 george meshtasticd[17641]: INFO  | ??:??:?? 11 [mqtt] MQTT connected
Jul 06 02:51:07 george meshtasticd[17641]: INFO  | ??:??:?? 11 [mqtt] published online=1
Jul 06 02:51:07 george meshtasticd[17641]: INFO  | ??:??:?? 12 [mqtt] Attempting to connect directly to MQTT server mqtt.meshtastic.org, port: 1883, username: meshdev, passw>
Jul 06 02:51:07 george meshtasticd[17641]: INFO  | ??:??:?? 12 [mqtt] MQTT connected
Jul 06 02:51:07 george meshtasticd[17641]: INFO  | ??:??:?? 12 [mqtt] published online=1
Jul 06 02:51:07 george meshtasticd[17641]: INFO  | ??:??:?? 12 [mqtt] Attempting to connect directly to MQTT server mqtt.meshtastic.org, port: 1883, username: meshdev, passw>
Jul 06 02:51:07 george meshtasticd[17641]: INFO  | ??:??:?? 12 [mqtt] MQTT connected
Jul 06 02:51:07 george meshtasticd[17641]: INFO  | ??:??:?? 12 [mqtt] published online=1
Jul 06 02:51:07 george meshtasticd[17641]: INFO  | ??:??:?? 13 [mqtt] Attempting to connect directly to MQTT server mqtt.meshtastic.org, port: 1883, username: meshdev, passw>
Jul 06 02:51:07 george meshtasticd[17641]: INFO  | ??:??:?? 13 [mqtt] MQTT connected
Jul 06 02:51:07 george meshtasticd[17641]: INFO  | ??:??:?? 13 [mqtt] published online=1
Jul 06 02:51:07 george meshtasticd[17641]: INFO  | ??:??:?? 13 [mqtt] Attempting to connect directly to MQTT server mqtt.meshtastic.org, port: 1883, username: meshdev, passw>
Jul 06 02:51:07 george meshtasticd[17641]: INFO  | ??:??:?? 13 [mqtt] MQTT connected
Jul 06 02:51:07 george meshtasticd[17641]: INFO  | ??:??:?? 13 [mqtt] published online=1
Jul 06 02:51:07 george meshtasticd[17641]: INFO  | ??:??:?? 13 [mqtt] Attempting to connect directly to MQTT server mqtt.meshtastic.org, port: 1883, username: meshdev, passw>
Jul 06 02:51:07 george meshtasticd[17641]: INFO  | ??:??:?? 13 [mqtt] MQTT connected
Jul 06 02:51:07 george meshtasticd[17641]: INFO  | ??:??:?? 13 [mqtt] published online=1
Jul 06 02:51:07 george meshtasticd[17641]: INFO  | ??:??:?? 13 [mqtt] Attempting to connect directly to MQTT server mqtt.meshtastic.org, port: 1883, username: meshdev, passw>
Jul 06 02:51:07 george meshtasticd[17641]: INFO  | ??:??:?? 13 [mqtt] MQTT connected
Jul 06 02:51:07 george meshtasticd[17641]: INFO  | ??:??:?? 13 [mqtt] published online=1
Jul 06 02:51:07 george meshtasticd[17641]: INFO  | ??:??:?? 14 [mqtt] Attempting to connect directly to MQTT server mqtt.meshtastic.org, port: 1883, username: meshdev, passw>
Jul 06 02:51:07 george meshtasticd[17641]: INFO  | ??:??:?? 14 [mqtt] MQTT connected
Jul 06 02:51:07 george meshtasticd[17641]: INFO  | ??:??:?? 14 [mqtt] published online=1
Jul 06 02:51:07 george meshtasticd[17641]: INFO  | ??:??:?? 14 [mqtt] Attempting to connect directly to MQTT server mqtt.meshtastic.org, port: 1883, username: meshdev, passw>
```

blu006, 2024-07-06:
If needed, I can also provide the schematic and design files between the OPi one and my RFM95, but I don't think those are necessary for this software issue.

jp-bennett, 2024-07-06:
Looks like pubSub.loop() at https://github.com/meshtastic/firmware/blob/master/src/mqtt/MQTT.cpp#L413 is always returning false, even though it should be connected. Very possibly an issue with Portduino's fake WiFiClient implementation.

blu006, 2024-07-06:
I guess I could set up a vscode environment with platform io, place a debug statement, and try to crosscompile for armhf...

jp-bennett, 2024-07-06:
@blu006 You could open a PR with some reasonable debug calls added. The CI will push out armhf and arm64 builds.

blu006, 2024-07-06:
@jp-bennett it seems to be as you expected (not surprised)

```
Jul 06 15:13:55 george meshtasticd[25790]: 
Jul 06 15:13:55 george meshtasticd[25790]: INFO  | ??:??:?? 96 [mqtt] 
Jul 06 15:13:55 george meshtasticd[25790]: MQTT connected
Jul 06 15:13:55 george meshtasticd[25790]: 
Jul 06 15:13:55 george meshtasticd[25790]: 
Jul 06 15:13:55 george meshtasticd[25790]: INFO  | ??:??:?? 96 [mqtt] 
Jul 06 15:13:55 george meshtasticd[25790]: published online=1
Jul 06 15:13:55 george meshtasticd[25790]: 
Jul 06 15:13:55 george meshtasticd[25790]: 
Jul 06 15:13:55 george meshtasticd[25790]: DEBUG | ??:??:?? 97 [mqtt] 
Jul 06 15:13:55 george meshtasticd[25790]: Reconnecting MQTT because the PubSub loop returned false
Jul 06 15:13:55 george meshtasticd[25790]: 
Jul 06 15:13:55 george meshtasticd[25790]: 
Jul 06 15:13:55 george meshtasticd[25790]: INFO  | ??:??:?? 97 [mqtt] 
Jul 06 15:13:55 george meshtasticd[25790]: Attempting to connect directly to MQTT server mqtt.meshtastic.org, port: 1883, username: meshdev, password: large4cats
Jul 06 15:13:55 george meshtasticd[25790]: 
Jul 06 15:13:55 george meshtasticd[25790]: 
Jul 06 15:13:55 george meshtasticd[25790]: INFO  | ??:??:?? 97 [mqtt] 
Jul 06 15:13:55 george meshtasticd[25790]: MQTT connected
Jul 06 15:13:55 george meshtasticd[25790]: 
Jul 06 15:13:55 george meshtasticd[25790]: 
Jul 06 15:13:55 george meshtasticd[25790]: INFO  | ??:??:?? 97 [mqtt] 
Jul 06 15:13:55 george meshtasticd[25790]: published online=1
Jul 06 15:13:55 george meshtasticd[25790]: 
Jul 06 15:13:55 george meshtasticd[25790]: 
Jul 06 15:13:55 george meshtasticd[25790]: DEBUG | ??:??:?? 97 [mqtt] 
Jul 06 15:13:55 george meshtasticd[25790]: Reconnecting MQTT because the PubSub loop returned false
Jul 06 15:13:55 george meshtasticd[25790]: 
Jul 06 15:13:55 george meshtasticd[25790]: 
Jul 06 15:13:55 george meshtasticd[25790]: INFO  | ??:??:?? 97 [mqtt] 
Jul 06 15:13:55 george meshtasticd[25790]: Attempting to connect directly to MQTT server mqtt.meshtastic.org, port: 1883, username: meshdev, password: large4cats
Jul 06 15:13:55 george meshtasticd[25790]: 
Jul 06 15:13:55 george meshtasticd[25790]: 
Jul 06 15:13:55 george meshtasticd[25790]: INFO  | ??:??:?? 97 [mqtt] 
Jul 06 15:13:55 george meshtasticd[25790]: MQTT connected
Jul 06 15:13:55 george meshtasticd[25790]: 
Jul 06 15:13:55 george meshtasticd[25790]: 
Jul 06 15:13:55 george meshtasticd[25790]: INFO  | ??:??:?? 97 [mqtt] 
Jul 06 15:13:55 george meshtasticd[25790]: published online=1
Jul 06 15:13:55 george meshtasticd[25790]: 
```

jp-bennett, 2024-07-06:
@blu006 We might be able to query pubsub's state() call to get the reason, as per https://pubsubclient.knolleary.net/api#state

blu006, 2024-07-06:
@jp-bennett  I get some interesting results from this.  It still looks like it's definitely a fake wifi issue.  I'm wondering how the portduino code for all of this looks. 

```
Jul 06 16:29:58 george meshtasticd[26864]: 
Jul 06 16:29:58 george meshtasticd[26864]: DEBUG | ??:??:?? 20 [mqtt] 
Jul 06 16:29:58 george meshtasticd[26864]: Reconnecting MQTT because the PubSub loop returned false:-3
Jul 06 16:29:58 george meshtasticd[26864]: 
Jul 06 16:29:58 george meshtasticd[26864]: 
Jul 06 16:29:58 george meshtasticd[26864]: INFO  | ??:??:?? 20 [mqtt] 
Jul 06 16:29:58 george meshtasticd[26864]: Attempting to connect directly to MQTT server mqtt.meshtastic.org, port: 1883, username: meshdev, password: large4cats
Jul 06 16:29:58 george meshtasticd[26864]: 
Jul 06 16:29:58 george meshtasticd[26864]: 
Jul 06 16:29:58 george meshtasticd[26864]: INFO  | ??:??:?? 20 [mqtt] 
Jul 06 16:29:58 george meshtasticd[26864]: MQTT connected
Jul 06 16:29:58 george meshtasticd[26864]: 
Jul 06 16:29:58 george meshtasticd[26864]: 
Jul 06 16:29:58 george meshtasticd[26864]: INFO  | ??:??:?? 20 [mqtt] 
Jul 06 16:29:58 george meshtasticd[26864]: published online=1
Jul 06 16:29:58 george meshtasticd[26864]: 
Jul 06 16:29:58 george meshtasticd[26864]: 
Jul 06 16:29:58 george meshtasticd[26864]: DEBUG | ??:??:?? 21 [mqtt] 
Jul 06 16:29:58 george meshtasticd[26864]: Reconnecting MQTT because the PubSub loop returned false:-3
Jul 06 16:29:58 george meshtasticd[26864]: 
Jul 06 16:29:58 george meshtasticd[26864]: 
Jul 06 16:29:58 george meshtasticd[26864]: INFO  | ??:??:?? 21 [mqtt] 
Jul 06 16:29:58 george meshtasticd[26864]: Attempting to connect directly to MQTT server mqtt.meshtastic.org, port: 1883, username: meshdev, password: large4cats
Jul 06 16:29:58 george meshtasticd[26864]: 
Jul 06 16:29:58 george meshtasticd[26864]: 
Jul 06 16:29:58 george meshtasticd[26864]: INFO  | ??:??:?? 21 [mqtt] 
Jul 06 16:29:58 george meshtasticd[26864]: MQTT connected
Jul 06 16:29:58 george meshtasticd[26864]: 
Jul 06 16:29:58 george meshtasticd[26864]: 
Jul 06 16:29:58 george meshtasticd[26864]: INFO  | ??:??:?? 21 [mqtt] 
Jul 06 16:29:58 george meshtasticd[26864]: published online=1
Jul 06 16:29:58 george meshtasticd[26864]: 
```

Talie5in, 2024-07-07:
@blu006 Part of your Issue says only Android can configure. 

Is this meaning iOS App can't?
What about the web client?

blu006, 2024-07-07:
> @blu006 Part of your Issue says only Android can configure. 
> 
> Is this meaning iOS App can't?
> What about the web client?

I could not configure it on the web client.

I do not have iOS

jp-bennett, 2024-07-08:
@blu006 It looks to me like https://github.com/knolleary/pubsubclient/blob/2d228f2f862a95846c65a8518c79f48dfc8f188c/src/PubSubClient.cpp#L684 calls https://github.com/meshtastic/WiFi/blob/b885b9595d54ee6eae59696eeae98f631eb27a23/src/WiFiClient.cpp#L171 which is the spot in the fake wifi code that is returning not connected.

blu006, 2024-07-08:
@jp-bennett I'm still looking for some way to cross-compile everything without the limitations from GitHub actions 😵‍💫

Is there a docker image that I can run under qemu or anything like that?

jp-bennett, 2024-07-08:
> @jp-bennett I'm still looking for some way to cross-compile everything without the limitations from GitHub actions 😵‍💫
> 
> Is there a docker image that I can run under qemu or anything like that

I'm sure there are ways. I've just been compiling on an arm32 install on a modern Pi. I'll see about getting a test build for you to try.

blu006, 2024-07-08:
Thank you thank you!

blu006, 2024-07-09:
I'm interested.

jp-bennett, 2024-07-10:
@blu006 Sorry for the delay. See if https://drive.google.com/file/d/16uXRGjITWo6D2dblD1LBiAHtIKFjiDFx/view?usp=sharing will work.

blu006, 2024-07-10:
> @blu006 Sorry for the delay. See if https://drive.google.com/file/d/16uXRGjITWo6D2dblD1LBiAHtIKFjiDFx/view?usp=sharing will work.

it still connection cycles, but it seems slower this time (or my wifi sucks more).  Let me see if I can get the logs.
```
Jul 10 03:30:27 george meshtasticd[1884]: 
Jul 10 03:30:27 george meshtasticd[1884]: INFO  | ??:??:?? 240 [mqtt] 
Jul 10 03:30:27 george meshtasticd[1884]: published online=1
Jul 10 03:30:27 george meshtasticd[1884]: 
Jul 10 03:30:27 george meshtasticd[1884]: 
Jul 10 03:30:27 george meshtasticd[1884]: INFO  | ??:??:?? 270 [mqtt] 
Jul 10 03:30:27 george meshtasticd[1884]: Attempting to connect directly to MQTT server mqtt.meshtastic.org, port: 1883, username: meshdev, password: large4cats
Jul 10 03:30:27 george meshtasticd[1884]: 
Jul 10 03:30:27 george meshtasticd[1884]: 
Jul 10 03:30:27 george meshtasticd[1884]: INFO  | ??:??:?? 270 [mqtt] 
Jul 10 03:30:27 george meshtasticd[1884]: MQTT connected
Jul 10 03:30:27 george meshtasticd[1884]: 
Jul 10 03:30:27 george meshtasticd[1884]: 
Jul 10 03:30:27 george meshtasticd[1884]: INFO  | ??:??:?? 270 [mqtt] 
Jul 10 03:30:27 george meshtasticd[1884]: published online=1
Jul 10 03:30:27 george meshtasticd[1884]: 
Jul 10 03:30:27 george meshtasticd[1884]: 
Jul 10 03:30:27 george meshtasticd[1884]: INFO  | ??:??:?? 285 [DeviceTelemetryModule] 
Jul 10 03:30:27 george meshtasticd[1884]: (Sending): air_util_tx=0.082417, channel_utilization=0.000000, battery_level=101, voltage=0.000000, uptime=285
Jul 10 03:30:27 george meshtasticd[1884]: 
Jul 10 03:30:27 george meshtasticd[1884]: 
Jul 10 03:30:27 george meshtasticd[1884]: INFO  | ??:??:?? 285 [DeviceTelemetryModule] 
Jul 10 03:30:27 george meshtasticd[1884]: Sending packet to phone
Jul 10 03:30:27 george meshtasticd[1884]: 
Jul 10 03:30:27 george meshtasticd[1884]: 
Jul 10 03:30:27 george meshtasticd[1884]: INFO  | ??:??:?? 285 
Jul 10 03:30:27 george meshtasticd[1884]: Telling client we have new packets 14
Jul 10 03:30:27 george meshtasticd[1884]: 
Jul 10 03:30:27 george meshtasticd[1884]: 
Jul 10 03:30:27 george meshtasticd[1884]: INFO  | ??:??:?? 285 [ServerAPI] 
Jul 10 03:30:27 george meshtasticd[1884]: getFromRadio=STATE_SEND_PACKETS
Jul 10 03:30:27 george meshtasticd[1884]: 
Jul 10 03:30:27 george meshtasticd[1884]: 
\Jul 10 03:30:27 george meshtasticd[1884]: INFO  | ??:??:?? 300 [mqtt] 
Jul 10 03:30:27 george meshtasticd[1884]: Attempting to connect directly to MQTT server mqtt.meshtastic.org, port: 1883, username: meshdev, password: large4cats
Jul 10 03:30:27 george meshtasticd[1884]: 
Jul 10 03:30:27 george meshtasticd[1884]: 
Jul 10 03:30:27 george meshtasticd[1884]: INFO  | ??:??:?? 300 [mqtt] 
Jul 10 03:30:27 george meshtasticd[1884]: MQTT connected
Jul 10 03:30:27 george meshtasticd[1884]: 
Jul 10 03:30:27 george meshtasticd[1884]: 
Jul 10 03:30:27 george meshtasticd[1884]: INFO  | ??:??:?? 300 [mqtt] 
Jul 10 03:30:27 george meshtasticd[1884]: published online=1
Jul 10 03:30:27 george meshtasticd[1884]: 
Jul 10 03:30:27 george meshtasticd[1884]: 
Jul 10 03:30:27 george meshtasticd[1884]: INFO  | ??:??:?? 330 [mqtt] 
Jul 10 03:30:27 george meshtasticd[1884]: Attempting to connect directly to MQTT server mqtt.meshtastic.org, port: 1883, username: meshdev, password: large4cats
Jul 10 03:30:27 george meshtasticd[1884]: 
Jul 10 03:30:27 george meshtasticd[1884]: 
Jul 10 03:30:27 george meshtasticd[1884]: INFO  | ??:??:?? 330 [mqtt] 
Jul 10 03:30:27 george meshtasticd[1884]: MQTT connected
Jul 10 03:30:27 george meshtasticd[1884]: 
Jul 10 03:30:27 george meshtasticd[1884]: 
Jul 10 03:30:27 george meshtasticd[1884]: INFO  | ??:??:?? 330 [mqtt] 
Jul 10 03:30:27 george meshtasticd[1884]: published online=1
Jul 10 03:30:27 george meshtasticd[1884]: 
```

blu006, 2024-07-12:
@jp-bennett are the debugging symbols retained in these pio files?  they seem a bit small for that imo

i kind of want to put a breakpoint in it with gdb and see how things go

closed 2024-10-06 by caveman99