meshtastic/firmware #4250: [Bug]: Restarting on Portduino / Linux Leaves Hanging API Ports

blu006, 2024-07-07:
### Category

Other

### Hardware

DIY

### Firmware Version

2.3.13.83f5ba0

### Description

The portduino server leaves hanging ports with the status `CLOSE_WAIT` when monitored through `lsof`.

My device is an orangepione.

`armbianmonitor -u` reports the following:
```

### lsusb:

Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=musb-hdrc/1p, 480M
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=ohci-platform/1p, 12M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=ohci-platform/1p, 12M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/1p, 480M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/1p, 480M

### Group membership of brandon : brandon tty disk dialout sudo audio video plugdev games users systemd-journal input render netdev

### Userland generated with Armbian Build Framework:

PRETTY_NAME="Armbian 24.5.1 bookworm"
ARMBIAN_PRETTY_NAME="Armbian 24.5.1 bookworm"

### Installed packages:

ii  armbian-bsp-cli-orangepione-current 24.5.1                         armhf        Armbian CLI BSP for board 'orangepione' branch 'current'
ii  armbian-firmware                    24.5.1                         all          Armbian - Linux firmware
ii  linux-base                          4.9                            all          Linux image base package
ii  linux-dtb-current-sunxi             24.5.1                         armhf        Armbian Linux current DTBs in /boot/dtb-6.6.31-current-sunxi
ii  linux-image-current-sunxi           24.5.1                         armhf        Armbian Linux current kernel image 6.6.31-current-sunxi
ii  linux-libc-dev:armhf                6.1.94-1                       armhf        Linux support headers for userspace development
ii  linux-u-boot-orangepione-current    24.5.1                         armhf        Das U-Boot for orangepione

### Loaded modules:

Module                  Size  Used by
pps_ldisc              16384  2
bluetooth             528384  10
ecdh_generic           12288  1 bluetooth
ecc                    24576  1 ecdh_generic
rfkill                 20480  2 bluetooth
snd_soc_hdmi_codec     16384  1
lz4hc                  12288  0
sunxi_cedrus           40960  0
v4l2_mem2mem           20480  1 sunxi_cedrus
sun9i_hdmi_audio       12288  0
sun4i_gpadc_iio        12288  0
lima                   49152  0
lz4                    12288  0
videobuf2_dma_contig    16384  1 sunxi_cedrus
industrialio           57344  1 sun4i_gpadc_iio
[... 608 lines cut ...]
Jul 06 00:11:42 george meshtasticd[11006]: INFO  | ??:??:?? 0 Loaded saved config version 22
Jul 06 00:11:42 george meshtasticd[11006]: INFO  | ??:??:?? 0 Loading /prefs/module.proto
Jul 06 00:11:42 george meshtasticd[1i1006]: INFO  | ??:??:?? 0 Loaded /prefs/module.proto successfully
Jul 06 00:11:42 george meshtasticd[11006]: INFO  | ??:??:?? 0 Loaded saved moduleConfig version 22
Jul 06 00:11:42 george meshtasticd[11006]: INFO  | ??:??:?? 0 Loading /prefs/channels.proto
Jul 06 00:11:42 george meshtasticd[11006]: INFO  | ??:??:?? 0 Loaded /prefs/channels.proto successfully
Jul 06 00:11:42 george meshtasticd[11006]: INFO  | ??:??:?? 0 Loaded saved channelFile version 22
Jul 06 00:11:42 george meshtasticd[11006]: INFO  | ??:??:?? 0 File /oem/oem.proto not found
Jul 06 00:11:42 george meshtasticd[11006]: INFO  | ??:??:?? 0 Wanted region 1, using US
``` 
Here is what `lsof -i 4 | grep mesh` outputs:
```
meshtasti 16997            root   15u  IPv4 5428322      0t0  TCP george.maglab:4403->wireguard.maglab:37912 (CLOSE_WAIT)
meshtasti 16997            root   18u  IPv4 5472319      0t0  TCP george.maglab:4403->wireguard.maglab:45012 (CLOSE_WAIT)
meshtasti 16997            root   21u  IPv4 5473526      0t0  TCP george.maglab:4403->wireguard.maglab:53472 (CLOSE_WAIT)
meshtasti 16997            root   24u  IPv4 5475343      0t0  TCP george.maglab:4403->wireguard.maglab:43764 (CLOSE_WAIT)
meshtasti 16997            root   26u  IPv4 5477977      0t0  TCP *:https (LISTEN)
meshtasti 16997            root   27u  IPv4 5477669      0t0  TCP george.maglab:4403->wireguard.maglab:34864 (CLOSE_WAIT)
meshtasti 16997            root   28u  IPv4 5477978      0t0  TCP *:4403 (LISTEN)
meshtasti 16997            root   29u  IPv4 5476786      0t0  TCP george.maglab:https->wireguard.maglab:35746 (ESTABLISHED)
meshtasti 16997            root   30u  IPv4 5488752      0t0  TCP george.maglab:4403->wireguard.maglab:41544 (ESTABLISHED)
```
Note: I am interfacing with this node remotely using wireguard.

I am unsure if it executes the code at https://github.com/meshtastic/firmware/blob/v2.3.13.83f5ba0/src/shutdown.h#L23 or not.

caveman99, 2024-12-20:
already assigned, so leave this open

fifieldt, 2025-10-08:
Is this still seen in a 2.7x version?