meshtastic/firmware #4256: [Bug]: loss of location

jranma, 2024-07-08:
### Category

Other

### Hardware

T-Beam S3

### Firmware Version

2.3.13

### Description

I set a "fixed" location, but sometimes it gets lost for no reason and settles on this location: 

Latitude: 0.0032768

Longitude: 0.0032768

Altitude: ???

this area is also populated by other nodes, so apparently I'm not the only one to experience this bug. 

the funny thing is that each one has a slightly different location, and lattitude and longitude always seem to be identical for each node. 

I haven't changed any settings and this bug seems to appear after disconnecting and reconnecting my router's power supply several times. Other settings are not affected.

ianmcorvidae, 2024-07-08:
I had my node set a position like this earlier today as well; happened when I did a reset of the NodeDB. My node is also set to use position precision of 16 on its primary channel (the default LongFast), and I was wondering if it could be related -- is your node using a limited precision of any kind?

nicheath, 2024-09-11:
> I had my node set a position like this earlier today as well; happened when I did a reset of the NodeDB. My node is also set to use position precision of 16 on its primary channel (the default [...]

[The clearing of position data occurs on a NodeDB reset](https://github.com/meshtastic/firmware/issues/3929). Why clear configuration data when clearing runtime/dynamic data like node list? Good question.

closed 2024-10-07 by thebentern