meshtastic/firmware #4268: [Feature Request]: Allow Battery ADC Pin override

jhbruhn, 2024-07-11:
### Platform

ESP32

### Description

For "DIY" Builds, like in my case the T-Lora v1 (I know...found it in my parts bin), an external battery solution might be built. For that, it would be great if the Battery ADC Pin could be overriden like the buzzer pin etc.

Is there anything that has to be kept in mind if I try to implement this functionality, or would that be easily possible?

InspiringCode, 2025-05-21:
But this means that overriding BATTERY_PIN means changing the source and manually recompiling, correct?

Did you consider adding an option to the UI like "ADC multiplier override ratio" but for the pin? I think that's a really common scenario for custom builds. Many people want/need to use external chargers or step-up converters (e.g. for sodium ion or 12v batteries).

This would be really appreciated!

Mictronics, 2025-05-21:
Rebuilding the source is required when when your change the #define related to the battery pin or ADC.

Mictronics, 2025-05-21:
Depends. I have no experience with how widely that is used. Changed my battery pin only for one variant. And since I build all firmware myself it's useless efforts IMHO.