meshtastic/firmware #4275: Critical fault #11[Bug]:

MonstaB, 2024-07-12:
### Category

Other

### Hardware

T-Deck

### Firmware Version

latest bata

### Description

can trigger this at will by either sending a pvt message that is max character length or by typing characters in a single word too long to fit on the screen without a space eg.. repeating letter KKKKKKKKKKKK all the way accross.
tell me how to get a log and ill do it

Nestpebble, 2024-11-24:
@MonstaB are you still able to reproduce this fault?

joyel24, 2024-12-15:
Testing on latest tft-gui-work commit [ca6a80a](https://github.com/meshtastic/firmware/commit/ca6a80ad6bf4047ab2f4fbf920fb1034c5ce3aca)
 
I tried to reproduce and got this bug 2 times.
Then started taking a video to catch it, but it never happened again during 8+min of capturing...

GUVWAF, 2024-12-16:
@joyel24 Please don't distract this issue from issues with the `tft-gui-work` branch, which is still WIP and has known issues with the shared SPI of the screen and radio.

joyel24, 2024-12-16:
SERIAL OUTPUT FROM USB
```shell
[0m [32m INFO  [0m| 14:13:55 95 [32m Tell client we have new packets 32
[0m [32m INFO  [0m| 14:13:55 95 [32m Tell client we have new packets 32
[0m [34m DEBUG [0m| 14:13:55 95 [PacketAPI] [34m FromRadio=STATE_SEND_PACKETS
[0m [34m DEBUG [0m| 14:13:55 95 [34m [DeviceUI] handleFromRadio variant 11
[0m [33m WARN  [0m| 14:13:55 95 [RadioIf] [33m Can not send yet, busyTx
[0m [34m DEBUG [0m| 14:13:55 95 [34m [DeviceUI] key press value: 110
[0m [33m WARN  [0m| 14:13:55 95 [RadioIf] [33m Can not send yet, busyTx
[0m [34m DEBUG [0m| 14:13:55 95 [RadioIf] [34m Completed sending (id=0x16f09b31 fr=0xa2ed5344 to=0xffffffff, WantAck=0, HopLim=3 Ch=0x8 encrypted rxtime=1734358434 hopStart=3 priority=100)
[0m [34m DEBUG [0m| 14:13:55 95 [RadioIf] [34m [DeviceUI] key press value: 13
[0m [34m DEBUG [0m| 14:13:55 95 [RadioIf] [34m [DeviceUI] sending meshpacket to radio id=0x7f48f734, to=0xffffffff(4294967295), ch=0, portnum=1, len=1, wantRsp=0
[0m [34m DEBUG [0m| 14:13:55 95 [PacketAPI] [34m Received packet id=20
[0m [34m DEBUG [0m| 14:13:55 95 [PacketAPI] [34m PACKET FROM QUEUE (id=0x7f48f734 fr=0x00000000 to=0xffffffff, WantAck=1, HopLim=3 Ch=0x0 Portnum=1)
[0m [34m DEBUG [0m| 14:13:55 95 [PacketAPI] [34m handleReceived(USER) (id=0x7f48f734 fr=0x00000000 to=0xffffffff, WantAck=1, HopLim=3 Ch=0x0 Portnum=1 rxtime=1734358435)
[0m [34m DEBUG [0m| 14:13:55 95 [PacketAPI] [34m Module 'text' wantsPacket=1
[0m [32m INFO  [0m| 14:13:55 95 [PacketAPI] [32m Received text msg from=0x0, id=0x7f48f734, msg=n
[0m [34m DEBUG [0m| 14:13:55 95 [PacketAPI] [34m State: ON
[0m [34m DEBUG [0m| 14:13:55 95 [PacketAPI] [34m Module 'text' considered
[0m [34m DEBUG [0m| 14:13:55 95 [PacketAPI] [34m Module 'StoreForward' wantsPacket=1
[0m [34m DEBUG [0m| 14:13:55 95 [PacketAPI] [34m Module 'StoreForward' considered
[0m [34m DEBUG [0m| 14:13:55 95 [PacketAPI] [34m Module 'ExternalNotificationModule' wantsPacket=1
[0m [34m DEBUG [0m| 14:13:55 95 [PacketAPI] [34m Module 'ExternalNotificationModule' considered
[0m [34m DEBUG [0m| 14:13:55 95 [PacketAPI] [34m Module 'routing' wantsPacket=1
[0m [32m INFO  [0m| 14:13:55 95 [PacketAPI] [32m Received routing from=0x0, id=0x7f48f734, portnum=1, payloadlen=1
[0m [34m DEBUG [0m| 14:13:55 95 [PacketAPI] [34m Routing sniffing (id=0x7f48f734 fr=0x00000000 to=0xffffffff, WantAck=1, HopLim=3 Ch=0x0 Portnum=1 rxtime=1734358435)
[0m [34m DEBUG [0m| 14:13:55 95 [PacketAPI] [34m Module 'routing' considered
[0m [34m DEBUG [0m| 14:13:55 95 [PacketAPI] [34m Set next retransmission in 7908 msecs: 
[0m [34m DEBUG [0m| 14:13:55 95 [PacketAPI] [34m  (id=0x7f48f734 fr=0x00000000 to=0xffffffff, WantAck=1, HopLim=3 Ch=0x0 Portnum=1 rxtime=1734358435)
[0m [34m DEBUG [0m| 14:13:55 95 [PacketAPI] [34m Add packet record (id=0x7f48f734 fr=0x00000000 to=0xffffffff, WantAck=1, HopLim=3 Ch=0x0 Portnum=1 rxtime=1734358435)
[0m [34m DEBUG [0m| 14:13:55 95 [PacketAPI] [34m Expand short PSK #1
[0m [34m DEBUG [0m| 14:13:55 95 [PacketAPI] [34m Use AES128 key!
[0m [34m DEBUG [0m| 14:13:55 95 [PacketAPI] [34m enqueue for send (id=0x7f48f734 fr=0xa2ed5344 to=0xffffffff, WantAck=0, HopLim=3 Ch=0x8 encrypted rxtime=1734358435 hopStart=3 priority=100)
[0m [34m DEBUG [0m| 14:13:55 95 [PacketAPI] [34m txGood=8,txRelay=0,rxGood=0,rxBad=0
[0m [32m INFO  [0m| 14:13:55 95 [32m Tell client we have new packets 33
[0m [32m INFO  [0m| 14:13:55 95 [32m Tell client we have new packets 33
[0m [34m DEBUG [0m| 14:13:55 95 [ServerAPI] [34m FromRadio=STATE_SEND_PACKETS
[0m [34m DEBUG [0m| 14:13:55 95 [RadioIf] [34m Started Tx (id=0xb5e36332 fr=0xa2ed5344 to=0xffffffff, WantAck=0, HopLim=3 Ch=0x8 encrypted rxtime=1734358434 hopStart=3 priority=100)
[0m [34m DEBUG [0m| 14:13:55 95 [RadioIf] [34m Packet TX: 403ms
[0m [34m DEBUG [0m| 14:13:55 95 [34m [DeviceUI] key press value: 110
[0m [34m DEBUG [0m| 14:13:55 95 [34m [DeviceUI] key press value: 13
[0m [34m DEBUG [0m| 14:13:55 95 [34m [DeviceUI] sending meshpacket to radio id=0xfe880335, to=0xffffffff(4294967295), ch=0, portnum=1, len=1, wantRsp=0
[0m [34m DEBUG [0m| 14:13:55 95 [PacketAPI] [34m Received packet id=21
[0m [34m DEBUG [0m| 14:13:55 95 [PacketAPI] [34m PACKET FROM QUEUE (id=0xfe880335 fr=0x00000000 to=0xffffffff, WantAck=1, HopLim=3 Ch=0x0 Portnum=1)
[0m [34m DEBUG [0m| 14:13:55 95 [PacketAPI] [34m handleReceived(USER) (id=0xfe880335 fr=0x00000000 to=0xffffffff, WantAck=1, HopLim=3 Ch=0x0 Portnum=1 rxtime=1734358435)
[0m [34m DEBUG [0m| 14:13:55 95 [PacketAPI] [34m Module 'text' wantsPacket=1
[0m [32m INFO  [0m| 14:13:55 95 [PacketAPI] [32m Received text msg from=0x0, id=0xfe880335, msg=n
[0m [34m DEBUG [0m| 14:13:55 95 [PacketAPI] [34m State: ON
[0m [34m DEBUG [0m| 14:13:55 95 [PacketAPI] [34m Module 'text' considered
[0m [34m DEBUG [0m| 14:13:55 95 [PacketAPI] [34m Module 'StoreForward' wantsPacket=1
[0m [34m DEBUG [0m| 14:13:55 95 [PacketAPI] [34m Module 'StoreForward' considered
[0m [34m DEBUG [0m| 14:13:55 96 [PacketAPI] [34m Module 'ExternalNotificationModule' wantsPacket=1
[0m [34m DEBUG [0m| 14:13:55 96 [PacketAPI] [34m Module 'ExternalNotificationModule' considered
[0m [34m DEBUG [0m| 14:13:55 96 [PacketAPI] [34m Module 'routing' wantsPacket=1
[... cut ...]
[0m [34m DEBUG [0m| 14:13:56 96 [RadioIf] [34m SX126x standby RadioLib err=-705
```

USART0 OUTPUT
```shell

assert failed: void SX126xInterface<T>::startReceive() [with T = SX1262] SX126xInterface.cpp:291 (err == RADIOLIB_ERR_NONE)

Backtrace: 0x40377f1a:0x3fced010 0x40381841:0x3fced030 0x40388dcd:0x3fced050 0x420658b8:0x3fced180 0x4206a8f7:0x3fced1a0 0x4206ab08:0x3fced1c0 0x42074a3e:0x3fced1e0 0x42074a4e:0x3fced200 0x42074a0e:0x3fced220 0x4206d461:0x3fced240 0x42019275:0x3fced280

ELF file SHA256: 4fea205f1d91495b

Rebooting...

```

GUVWAF, 2024-12-16:
As I said, there's a remaining issue with the new UI due to the shared SPI instance with the screen. This is unrelated to the original issue unless you can reproduce it with an official firmware release.

joyel24, 2024-12-16:
> As I said, there's a remaining issue with the new UI due to the shared SPI instance with the screen. This is unrelated to the original issue unless you can reproduce it with an official firmware release.

I don't know if the author of this issue is talking about the master latest bata or the tft ui branch...
I'll flash the master branch and see if it's also happening.

joyel24, 2024-12-16:
Well, just did some experiments with the actual latest release... 2.5.15.79da236

I encounter a annoying behavior: The message content has been erased due to a false LONG PRESS on touchscreen. 
I did't touch the screen at all while writing the msg with keyboard...

```shell
[0m [34m DEBUG [0m| ??:??:?? 94 [touchscreen1] [34m action LONG PRESS(284/174)
[0m [34m DEBUG [0m| ??:??:?? 94 [touchscreen1] [34m State: ON
[0m [34m DEBUG [0m| ??:??:?? 94 [touchscreen1] [34m Show standard frames
[0m [34m DEBUG [0m| ??:??:?? 94 [touchscreen1] [34m Show 0 module frames
[0m [34m DEBUG [0m| ??:??:?? 94 [touchscreen1] [34m Total frame count: 103
[0m [34m DEBUG [0m| ??:??:?? 94 [touchscreen1] [34m Added modules.  numframes: 0
[0m [34m DEBUG [0m| ??:??:?? 94 [touchscreen1] [34m Finished build frames. numframes: 3
```

But it's not a critical fault so propably no link with the author...
@MonstaB Can you give more Information ?

caveman99, 2024-12-16:
> I don't know if the author of this issue is talking about the master or the tft ui branch... I'll flash the master branch and see if it's also happening.

The TFT-UI branch is not part of the meshtastic firmware (yet). As long as it is a separate branch, we don't accept bug reports against it. Also this is not a collection thread for various issues you are having, please describe only one problem and try to reproduce it.

joyel24, 2024-12-16:
It can be closed.
 https://github.com/meshtastic/device-ui/issues/24#issuecomment-2546261679
 
Sorry, i was trying to clarify this issue that was lacking a lot of Informations... And it's now clarified ✅ 
I may open another issue in the right repo concerning the bug I got.

closed 2024-12-17 by garthvh

fifieldt, 2025-02-27:
There is a new firmware preview for T-Deck that is a significant improvement, if you have this issue I recommend trying it out: https://meshtastic.org/blog/meshtastic-2-6-preview/ !