meshtastic/firmware #4291: [Bug]: MQTT Server Address Length

kjake, 2024-07-15:
### Category

Other

### Hardware

Heltec V3

### Firmware Version

2.3.13

### Description

I'm unable to configure either via the iOS app or the python CLI a MQTT server address longer than around 30 characters, despite `char address[64];` being defined in the source.  Whenever putting in the full server address, it is ignored without error, so I progressively added more characters until it stopped taking any more and reached 30.  When using things like Amazon MQ, etc, the generated DNS names are huge - and they need to be used in order to validate TLS connections (i.e. I can't just use a shorter CNAME to get around it).  The issue exists in 2.3.13 and 2.3.15.

garthvh, 2024-07-15:
62 bytes is the the practical limit for the field, there is not much space left on the mqtt config message.

kjake, 2024-07-15:
Ok. At least in my example, I am trying to store 34 characters. My
understanding is that each character would take a byte?

garthvh, 2024-07-17:
The iOS part of this is fixed and in TestFlight @ianmcorvidae is there a reason python would be using 30? (This must have been an older value)

ianmcorvidae, 2024-07-17:
I don't know a reason it would be limiting it; it looks like the value in module_config.options was updated from 32 to 64 in January or so, but I'm not sure if the .options files even make it into anything code-wise. I'm also able to do it locally (with effectively 43 characters of noise, but anyway).

kjake, could you check your CLI version? I'm not sure if this might be something that's changed, if you're on an older version.

kjake, 2024-07-17:
Looks like I have the latest version installed.

```
 ➜  ~ meshtastic --version
2.3.14

➜  ~ pip3 install --upgrade meshtastic
Requirement already satisfied: meshtastic in ./Documents/meshtastic/lib/python3.12/site-packages (2.3.14)
Requirement already satisfied: bleak<0.22.0,>=0.21.1 in ./Documents/meshtastic/lib/python3.12/site-packages (from meshtastic) (0.21.1)
Requirement already satisfied: dotmap<2.0.0,>=1.3.30 in ./Documents/meshtastic/lib/python3.12/site-packages (from meshtastic) (1.3.30)
Requirement already satisfied: packaging<25.0,>=24.0 in ./Documents/meshtastic/lib/python3.12/site-packages (from meshtastic) (24.1)
Requirement already satisfied: pexpect<5.0.0,>=4.9.0 in ./Documents/meshtastic/lib/python3.12/site-packages (from meshtastic) (4.9.0)
Requirement already satisfied: print-color<0.5.0,>=0.4.6 in ./Documents/meshtastic/lib/python3.12/site-packages (from meshtastic) (0.4.6)
Requirement already satisfied: protobuf>=5.26.0 in ./Documents/meshtastic/lib/python3.12/site-packages (from meshtastic) (5.27.2)
Requirement already satisfied: pyparsing<4.0.0,>=3.1.2 in ./Documents/meshtastic/lib/python3.12/site-packages (from meshtastic) (3.1.2)
Requirement already satisfied: pypubsub<5.0.0,>=4.0.3 in ./Documents/meshtastic/lib/python3.12/site-packages (from meshtastic) (4.0.3)
Requirement already satisfied: pyqrcode<2.0.0,>=1.2.1 in ./Documents/meshtastic/lib/python3.12/site-packages (from meshtastic) (1.2.1)
Requirement already satisfied: pyserial<4.0,>=3.5 in ./Documents/meshtastic/lib/python3.12/site-packages (from meshtastic) (3.5)
Requirement already satisfied: pyyaml<7.0.0,>=6.0.1 in ./Documents/meshtastic/lib/python3.12/site-packages (from meshtastic) (6.0.1)
Requirement already satisfied: requests<3.0.0,>=2.31.0 in ./Documents/meshtastic/lib/python3.12/site-packages (from meshtastic) (2.32.3)
Requirement already satisfied: tabulate<0.10.0,>=0.9.0 in ./Documents/meshtastic/lib/python3.12/site-packages (from meshtastic) (0.9.0)
Requirement already satisfied: webencodings<0.6.0,>=0.5.1 in ./Documents/meshtastic/lib/python3.12/site-packages (from meshtastic) (0.5.1)
Requirement already satisfied: pyobjc-core<10.0,>=9.2 in ./Documents/meshtastic/lib/python3.12/site-packages (from bleak<0.22.0,>=0.21.1->meshtastic) (9.2)
Requirement already satisfied: pyobjc-framework-CoreBluetooth<10.0,>=9.2 in ./Documents/meshtastic/lib/python3.12/site-packages (from bleak<0.22.0,>=0.21.1->meshtastic) (9.2)
Requirement already satisfied: pyobjc-framework-libdispatch<10.0,>=9.2 in ./Documents/meshtastic/lib/python3.12/site-packages (from bleak<0.22.0,>=0.21.1->meshtastic) (9.2)
Requirement already satisfied: ptyprocess>=0.5 in ./Documents/meshtastic/lib/python3.12/site-packages (from pexpect<5.0.0,>=4.9.0->meshtastic) (0.7.0)
Requirement already satisfied: charset-normalizer<4,>=2 in ./Documents/meshtastic/lib/python3.12/site-packages (from requests<3.0.0,>=2.31.0->meshtastic) (3.3.2)
Requirement already satisfied: idna<4,>=2.5 in ./Documents/meshtastic/lib/python3.12/site-packages (from requests<3.0.0,>=2.31.0->meshtastic) (3.7)
Requirement already satisfied: urllib3<3,>=1.21.1 in ./Documents/meshtastic/lib/python3.12/site-packages (from requests<3.0.0,>=2.31.0->meshtastic) (2.2.2)
Requirement already satisfied: certifi>=2017.4.17 in ./Documents/meshtastic/lib/python3.12/site-packages (from requests<3.0.0,>=2.31.0->meshtastic) (2024.7.4)
Requirement already satisfied: pyobjc-framework-Cocoa>=9.2 in ./Documents/meshtastic/lib/python3.12/site-packages (from pyobjc-framework-CoreBluetooth<10.0,>=9.2->bleak<0.22.0,>=0.21.1->meshtastic) (9.2)
```

closed 2024-08-11 by garthvh