meshtastic/firmware #4293: [Bug]: Meshtastic: No GPS

LoRa-user, 2024-07-15:
### Category

Other

### Hardware

Other

### Firmware Version

2.3.13.83f5ba0

### Description

After upgrading Meshtastic to 2.3.13, the CanaryOne device stopped detecting the GPS module.  All settings seem to be OK. No setting changes or multiple reboots or factory reset restore position recognition. "No GPS" and "No GPS present" are shown onscreen.

It should be noted, that after unboxing the CanaryOne presented the position in the right way.

fifieldt, 2024-07-16:
Thanks for reporting. Is it possible for you to attach serial logs?

https://flasher.meshtastic.org/ --> Open Serial Monitor

ms74ro, 2024-07-16:
I just installed the GPS module and it doesn't work. It appears to me what the user above mentioned. I checked the connections several times. It runs version 2.3.15.

[meshtastic-log-2024-07-16T13-26-17.225Z.log](https://github.com/user-attachments/files/16249547/meshtastic-log-2024-07-16T13-26-17.225Z.log)

LoRa-user, 2024-07-17:
The log from Flasher is here:

[meshtastic-log-2024-07-17T10-06-49.626Z.log](https://github.com/user-attachments/files/16263827/meshtastic-log-2024-07-17T10-06-49.626Z.log)

As it is coded (encrypted?) I don't know if there are interesting informations. Anyway, I have some logs from Python CLI and I am ready to send them for further investigations.

GPSFan, 2024-07-29:
Switching the ground of the GPS module with an NPN transistor is NOT good engineering practice.IMHO the Meshtastic docs page should be changed to reflect this.
While it may "work" in some cases there are many reasons why power control of the GPS module should not be done this way.

fifieldt, 2024-09-14:
Hi @ms74ro @LoRa-user - there have been some updates to the GPS detection code that went in to the 2.5.0 alpha. Are you able to give it a try?

fifieldt, 2024-10-12:
Fixed by https://github.com/meshtastic/firmware/pull/5014 - Try 2.5.7 when released.

closed 2024-10-12 by fifieldt