meshtastic/firmware #4298: [Bug]:

orhunavcu, 2024-07-17:
### Category

Serial

### Hardware

Other

### Firmware Version

2.3.13.83f5ba0

### Description

Hi, I am encountering issues while setting up Meshtastic on my Raspberry Pi 5 with a Waveshare LoRa 433MHz GNSS module. Below is the log output and specific errors I am facing:

`Portduino is starting, HWID=1, VFS root at /root/.portduino/default
Setting up Meshtastic on Portduino...
Using /etc/meshtasticd/config.yaml as config file
Warning, cannot claim pin GPIO21St16invalid_argument
Warning, cannot claim pin GPIO16St16invalid_argument
Warning, cannot claim pin GPIO20St16invalid_argument
Warning, cannot claim pin GPIO18St16invalid_argument
INFO  | ??:??:?? 0 

//\ E S H T /\ S T / C

DEBUG | ??:??:?? 0 Filesystem files:
DEBUG | ??:??:?? 0  /.. (0 Bytes)
DEBUG | ??:??:?? 0  /prefs/.. (0 Bytes)
DEBUG | ??:??:?? 0  /prefs/db.proto (512 Bytes)
DEBUG | ??:??:?? 0  /prefs/channels.proto (57 Bytes)
DEBUG | ??:??:?? 0  /prefs/. (0 Bytes)
DEBUG | ??:??:?? 0  /. (0 Bytes)
INFO  | ??:??:?? 0 No I2C device configured, skipping.
INFO  | ??:??:?? 0 No I2C devices found
INFO  | ??:??:?? 0 Meshtastic hwvendor=37, swver=2.3.13.83f5ba0
INFO  | ??:??:?? 0 Initializing NodeDB
INFO  | ??:??:?? 0 Loading /prefs/db.proto
INFO  | ??:??:?? 0 Loaded /prefs/db.proto successfully
INFO  | ??:??:?? 0 Loaded saved devicestate version 22, with nodecount: 1
INFO  | ??:??:?? 0 File /prefs/config.proto not found
INFO  | ??:??:?? 0 Installing default LocalConfig
INFO  | ??:??:?? 0 Setting default channel and radio preferences!
DEBUG | ??:??:?? 0 Expanding short PSK #1
INFO  | ??:??:?? 0 Wanted region 0, using UNSET
INFO  | ??:??:?? 0 File /prefs/module.proto not found
INFO  | ??:??:?? 0 Installing default ModuleConfig
INFO  | ??:??:?? 0 Loading /prefs/channels.proto
INFO  | ??:??:?? 0 Loaded /prefs/channels.proto successfully
INFO  | ??:??:?? 0 Loaded saved channelFile version 22
INFO  | ??:??:?? 0 File /oem/oem.proto not found
DEBUG | ??:??:?? 0 cleanupMeshDB purged 0 entries
DEBUG | ??:??:?? 0 Using nodenum 0x6732e961 
DEBUG | ??:??:?? 0 Expanding short PSK #1
INFO  | ??:??:?? 0 Wanted region 0, using UNSET
DEBUG | ??:??:?? 0 Set Timezone to GMT0
DEBUG | ??:??:?? 0 Read RTC time as 1721170151
DEBUG | ??:??:?? 0 NeighborInfoModule is disabled
INFO  | ??:??:?? 0 File /prefs/cannedConf.proto not found
INFO  | ??:??:?? 0 CannedMessageModule is enabled
DEBUG | ??:??:?? 0 Attempting to activate sx1262 radio on SPI port /dev/spidev0.0
DEBUG | ??:??:?? 0 SX126xInterface(cs=-1, irq=-1, rst=-1, busy=-1)
DEBUG | ??:??:?? 0 SX126X_DIO3_TCXO_VOLTAGE not defined, not using DIO3 as TCXO reference voltage
INFO  | ??:??:?? 0 Starting meshradio init...
DEBUG | ??:??:?? 0 (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=0, time 231 ms
DEBUG | ??:??:?? 0 (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=253, time 2115 ms
INFO  | ??:??:?? 0 Radio freq=906.875, config.lora.frequency_offset=0.000
INFO  | ??:??:?? 0 Set radio: region=UNSET, name=LongFast, config=0, ch=19, power=30
INFO  | ??:??:?? 0 Radio myRegion->freqStart -> myRegion->freqEnd: 902.000000 -> 928.000000 (26.000000 mhz)
INFO  | ??:??:?? 0 Radio myRegion->numChannels: 104 x 250.000kHz
INFO  | ??:??:?? 0 Radio channel_num: 20
INFO  | ??:??:?? 0 Radio frequency: 906.875000
INFO  | ??:??:?? 0 Slot time: 42 msec
INFO  | ??:??:?? 0 Set radio: final power level=22
INFO  | ??:??:?? 10 SX126x init result -2
ERROR | ??:??:?? 10 Failed to find SX1262 radio`

The main issues are:

I am receiving warnings about being unable to claim several GPIO pins (GPIO21, GPIO16, GPIO20, GPIO18).
The SX1262 radio is not being detected, resulting in the error “Failed to find SX1262 radio.”
Could someone please help me troubleshoot these issues?

Thank you!

PS: I did the same steps in this video. [https://www.youtube.com/watch?v=91ULi9DWgds&t=1317s](url)

### Relevant log output

```Shell
Lora:
  Module: sx1262  # Waveshare SX126X XXXM
  DIO2_AS_RF_SWITCH: true
  CS: 21
  IRQ: 16
  Busy: 20
  Reset: 18

gpiochip: 4

GPS:
  SerialPath: /dev/ttyAMA10

Webserver:
  Port: 443 # Port for Webserver & Webservices
  RootPath: /usr/share/doc/meshtasticd/web # Root Dir of WebServer

JOURNALCTL 

Tem 17 22:26:28 raspberrypi meshtasticd[1430]: 
Tem 17 22:26:28 raspberrypi meshtasticd[1430]: 
Tem 17 22:26:28 raspberrypi meshtasticd[1430]: INFO  | ??:??:?? 0 
Tem 17 22:26:28 raspberrypi meshtasticd[1430]: Set RX gain to boosted mode; result: 0
Tem 17 22:26:28 raspberrypi meshtasticd[1430]: 
Tem 17 22:26:28 raspberrypi meshtasticd[1430]: 
Tem 17 22:26:28 raspberrypi meshtasticd[1430]: ERROR | ??:??:?? 0 
Tem 17 22:26:28 raspberrypi meshtasticd[1430]: Failed to find SX1262 radio
Tem 17 22:26:28 raspberrypi meshtasticd[1430]: 
Tem 17 22:26:28 raspberrypi systemd[1]: meshtasticd.service: Failed with result 'exit-code'.
```

orhunavcu, 2024-07-17:
Hi, thanks for quick reply.

I managed to start sudo meshtasticd

and now it looks like it boots, but;

`Portduino is starting, HWID=1, VFS root at /root/.portduino/default
Setting up Meshtastic on Portduino...
Using /etc/meshtasticd/config.yaml as config file
INFO  | ??:??:?? 0 

//\ E S H T /\ S T / C

DEBUG | ??:??:?? 0 Filesystem files:
DEBUG | ??:??:?? 0  /.. (0 Bytes)
DEBUG | ??:??:?? 0  /prefs/.. (0 Bytes)
DEBUG | ??:??:?? 0  /prefs/db.proto (512 Bytes)
DEBUG | ??:??:?? 0  /prefs/channels.proto (57 Bytes)
DEBUG | ??:??:?? 0  /prefs/. (0 Bytes)
DEBUG | ??:??:?? 0  /. (0 Bytes)
INFO  | ??:??:?? 0 No I2C device configured, skipping.
INFO  | ??:??:?? 0 No I2C devices found
INFO  | ??:??:?? 0 S:B:37,2.3.15.deb7c27
INFO  | ??:??:?? 0 Initializing NodeDB
INFO  | ??:??:?? 0 Loading /prefs/db.proto
INFO  | ??:??:?? 0 Loaded /prefs/db.proto successfully
INFO  | ??:??:?? 0 Loaded saved devicestate version 22, with nodecount: 1
INFO  | ??:??:?? 0 File /prefs/config.proto not found
INFO  | ??:??:?? 0 Installing default LocalConfig
INFO  | ??:??:?? 0 Setting default channel and radio preferences!
DEBUG | ??:??:?? 0 Expanding short PSK #1
INFO  | ??:??:?? 0 Wanted region 0, using UNSET
INFO  | ??:??:?? 0 File /prefs/module.proto not found
INFO  | ??:??:?? 0 Installing default ModuleConfig
INFO  | ??:??:?? 0 Loading /prefs/channels.proto
INFO  | ??:??:?? 0 Loaded /prefs/channels.proto successfully
INFO  | ??:??:?? 0 Loaded saved channelFile version 22
INFO  | ??:??:?? 0 File /oem/oem.proto not found
DEBUG | ??:??:?? 0 cleanupMeshDB purged 0 entries
DEBUG | ??:??:?? 0 Using nodenum 0x6732e961 
DEBUG | ??:??:?? 0 Expanding short PSK #1
INFO  | ??:??:?? 0 Wanted region 0, using UNSET
DEBUG | ??:??:?? 0 Set Timezone to GMT0
DEBUG | ??:??:?? 0 Read RTC time as 1721247079
DEBUG | ??:??:?? 0 NeighborInfoModule is disabled
INFO  | ??:??:?? 0 File /prefs/cannedConf.proto not found
INFO  | ??:??:?? 0 CannedMessageModule is enabled
DEBUG | ??:??:?? 0 Attempting to activate sx1262 radio on SPI port /dev/spidev0.0
DEBUG | ??:??:?? 0 SX126xInterface(cs=21, irq=16, rst=18, busy=20)
DEBUG | ??:??:?? 0 SX126X_DIO3_TCXO_VOLTAGE not defined, not using DIO3 as TCXO reference voltage
INFO  | ??:??:?? 0 Starting meshradio init...
DEBUG | ??:??:?? 0 (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=0, time 231 ms
DEBUG | ??:??:?? 0 (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=253, time 2115 ms
INFO  | ??:??:?? 0 Radio freq=906.875, config.lora.frequency_offset=0.000
INFO  | ??:??:?? 0 Set radio: region=UNSET, name=LongFast, config=0, ch=19, power=30
INFO  | ??:??:?? 0 Radio myRegion->freqStart -> myRegion->freqEnd: 902.000000 -> 928.000000 (26.000000 MHz)
INFO  | ??:??:?? 0 Radio myRegion->numChannels: 104 x 250.000kHz
INFO  | ??:??:?? 0 Radio channel_num: 20
INFO  | ??:??:?? 0 Radio frequency: 906.875000
INFO  | ??:??:?? 0 Slot time: 42 msec
INFO  | ??:??:?? 0 Set radio: final power level=22
INFO  | ??:??:?? 0 SX126x init result 0
INFO  | ??:??:?? 0 Frequency set to 906.875000
INFO  | ??:??:?? 0 Bandwidth set to 250.000000
INFO  | ??:??:?? 0 Power output set to 22
DEBUG | ??:??:?? 0 Current limit set to 140.000000
DEBUG | ??:??:?? 0 Current limit set result 0
DEBUG | ??:??:?? 0 Setting DIO2 as RF switch
DEBUG | ??:??:?? 0 Using MCU pin -1 as RXEN and pin -1 as TXEN to control RF switching
INFO  | ??:??:?? 0 Set RX gain to boosted mode; result: 0
[... 16 lines cut ...]
DEBUG | ??:??:?? 30 [NodeInfoModule] localSend to channel 0
DEBUG | ??:??:?? 30 [NodeInfoModule] Add packet record (id=0x6b8b4569 fr=0x61 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=4 WANTRESP priority=10)
DEBUG | ??:??:?? 30 [NodeInfoModule] Expanding short PSK #1
DEBUG | ??:??:?? 30 [NodeInfoModule] Using AES128 key!
DEBUG | ??:??:?? 30 [NodeInfoModule] Installing AES128 key!
WARN  | ??:??:?? 30 [NodeInfoModule] send - lora tx disable because RegionCode_Unset
INFO  | ??:??:?? 45 [DeviceTelemetryModule] (Sending): air_util_tx=0.000000, channel_utilization=0.000000, battery_level=101, voltage=0.000000, uptime=45
DEBUG | ??:??:?? 45 [DeviceTelemetryModule] updateTelemetry LOCAL
DEBUG | ??:??:?? 45 [DeviceTelemetryModule] Node status update: 0 online, 1 total
INFO  | ??:??:?? 45 [DeviceTelemetryModule] Sending packet to mesh
DEBUG | ??:??:?? 45 [DeviceTelemetryModule] Update DB node 0x6732e961, rx_time=0
DEBUG | ??:??:?? 45 [DeviceTelemetryModule] handleReceived(LOCAL) (id=0x6b8b456a fr=0x61 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=67 priority=10)
DEBUG | ??:??:?? 45 [DeviceTelemetryModule] No modules interested in portnum=67, src=LOCAL
DEBUG | ??:??:?? 45 [DeviceTelemetryModule] localSend to channel 0
DEBUG | ??:??:?? 45 [DeviceTelemetryModule] Add packet record (id=0x6b8b456a fr=0x61 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=67 priority=10)
DEBUG | ??:??:?? 45 [DeviceTelemetryModule] Expanding short PSK #1
DEBUG | ??:??:?? 45 [DeviceTelemetryModule] Using AES128 key!
DEBUG | ??:??:?? 45 [DeviceTelemetryModule] Installing AES128 key!
WARN  | ??:??:?? 45 [DeviceTelemetryModule] send - lora tx disable because RegionCode_Unset

`

When I go to browser and write the https://localhost or https://myname

It shows UNK device. Can't edit or save the setting on it.

closed 2024-07-17 by orhunavcu

MJLEVER00, 2025-01-13:
Hi Orhunavcu,
how did you solve this? It is not clear from the above and I have the exact same issue!

orhunavcu, 2025-01-14:
Sorry It's been a while since July I don't quitely remember it but I belive I reflashed the Pi OS on Pi 5 and installed the fresh Meshtastic on it. Mesh is buggy on Raspberrys. If it not work then try to install different version of Meshtastic. 

If you running it on Pi 5 make sure to select  " gpiochip: 4 "   in config.yaml file.

MJLEVER00, 2025-01-18:
Thank you.

Solved it, by (indeed) starting all over and using this video instead of the instructions on the Meshtastic website:
https://youtu.be/91ULi9DWgds?si=Cx-3LSn7mPbP92k8

orhunavcu, 2025-01-18:
Super! It is nice to see your update. I don't know what caused this issue but it is random :)

Have a Meshy days :)