meshtastic/firmware #4309: [Bug]: "emoji" short names crashing Heltec V3 in "client" mode

martinbogo, 2024-07-20:
### Category

Other

### Hardware

Heltec V3

### Firmware Version

2.3.15.deb7c27

### Description

On a deployed mesh, I have four nodes which use Heltec V3, all in client mode, OLED display enabled, MQTT disabled, GPS and all sensor-modes disabled for debugging.

The nodes had long names set to "Tanglewild Fire" "Tanglewild Strawberry" "Tanglewild TRK01" and "Tanglewild TRK02" respectively

The nodes had short names set to "🔥, 🍓, 📻, 👻" respectively.

I attempted to capture a debug log, but the crash is occurring somewhere before an output is created in debug.  I am currently attempting root-cause analysis by adding more debug code around areas where the short name is used.

I was getting rapid boot loops on all four nodes after updating to 2.3.15 which I was not getting in 2.3.14.  Changing the short names to "fire, strb, trk1, trk2" caused the nodes to continue operating normally.  Doing so required shutting down all but the node being updated so my assumption is that this crash is occurring somewhere during one of:

- construction of the node name for transmission
- reading of the node name from flash/ram
- transmission of the node name to another node
- display of a node name on the OLED display

### Relevant log output

```Shell
Gathering debug logs - will update once more I have implemented more debug code harnesses around short-name handling.
```

todd-herbert, 2024-07-21:
My first thought is https://github.com/meshtastic/firmware/pull/4134. I'll dig into it later today.

todd-herbert, 2024-07-21:
I'm having trouble replicating the issue here, although I don't have the exact same hardware for testing either.
Two quick thoughts:
* are you building with either the `OLED_UA` or `OLED_RU` flag?
* have you tried erasing the flash before installing 2.3.15?

martinbogo, 2024-07-21:
> exact same hardware for testing either, sorry.

Is erasing the flash a prerequisite for 2.3.15?   I will do so.

todd-herbert, 2024-07-22:
> Is erasing the flash a prerequisite for 2.3.15? I will do so.

It really shouldn't be, it's just one of the first things I try myself if I get weird behaviour. Although, the fact that you're seeing the same issue on multiple nodes does make it sound more like a firmware bug.

I'm very interested to hear if you find any clues!

ravenium, 2024-07-22:
My short name is 🤔 and I haven't noticed any crashes. Running .15 on a heltec v3.  Pretty vanilla otherwise but willing to help if I can!

martinbogo, 2024-07-23:
Erasing the flash prior to upgrading to 2.3.15 does seem to have alleviated the issue on the Heltec V3 I just tried.  Will continue to test and get back on this.

fifieldt, 2024-09-15:
Hi @martinbogo . Have you had time to look into this further? Otherwise, since yours is working now, maybe we should close this :)

martinbogo, 2024-09-15:
Yeah definitely close it.

closed 2024-09-15 by garthvh