meshtastic/firmware #4318: 🐞 [Bug]: Locally connected node reports Hop Count of 1

RPN707, 2024-07-23:
### Firmware Version

2.3.13

### What did you do?

Updated FW to 2.3.13. Allowed iOS to update app to 2.3.11 (1025)

### Expected Behavior

Would expect node connected to via Bluetooth to not report a hop count

### Current Behavior

iOS 17.5.1 - FW 2.3.13 - App 2.3.11 (1025)
Tried 3 nodes, T-Echo, RAK 4631, Heltec V3. All same behavior.
Uninstalled and reinstalled app.
Since updating app to 2.3.11, node shows "rabbit" hop count of 1 when connected via Bluetooth in the node list.

[image]

### Participation

- [ ] I am willing to submit a pull request for this issue.

fifieldt, 2024-09-16:
@RPN707 , do you still see this in the latest firmware?

thebentern, 2024-09-16:
This was fixed in #4618

closed 2024-09-16 by thebentern

RPN707, 2024-09-17:
I'm only on 2.4.2 but, yes, I do still see a hop count on a BT connected node as of this morning.