meshtastic/firmware #4327: [Bug]: RAKwireless GNSS GPS Location Module u-blox ZOE-M8Q crashes firmware when not in Slot A

djekl, 2024-07-24:
### Category

Hardware Compatibility

### Hardware

Rak4631

### Firmware Version

2.3.14+

### Description

I recently purchased the `RAKwireless GNSS GPS Location Module u-blox ZOE-M8Q` for my `RAK19003 Base Board`. As per the RAK documentation, this should be supported in both expansion slots A and D.

```
Four Sensor Slots: The sensor slots A to D are used to connect with the I2C bus.
Slot D can be used for GNSS modules, too.
```

https://docs.rakwireless.com/Product-Categories/WisBlock/RAK19007/Datasheet/#overview
[image]

However, I was noticing that this GNSS module was crashing both of my 2 RAK nodes when connected and it was able to get a GPS lock.

Upon further reading, I was able to find this in the Meshtastic Documentation - https://meshtastic.org/docs/hardware/devices/rak/peripherals/?rakmodules=GPS#rak12500

This states slot A only.

Why only Slot A when the clear intention from the manufacturer is for it to go into slot D on the back, as per the longer silkscreen and hardware documentation as linked above. When placed in Slot A, it also covers Slot B, and reduces the number of expansion slots to only an additional 2 rather than 3.

garthvh, 2024-07-24:
You need to change the user button GPIO to use a slot other than a

djekl, 2024-07-24:
I have no GPIO enabled, or connected. Just the GPS and a Temp/Humid sensor from RAK.

djekl, 2024-07-24:
If it works on other ports (with caveats) can the docs be updated?

garthvh, 2024-07-24:
Io5 is the default user button, feel free to update the docs, there is an edit link on every page.

djekl, 2024-07-24:
@garthvh Help me out here please, and just explain things a little more.

So in the Meshtastic App (iOS) everything is GPIO pins.
- Under Device Config, both `Button GPIO` and `Buzzer GPIO` are `Unset`
- Under Canned Messages, again `Pin A`, `Pin B`, and `Press Pin` are all `Unset`
- Under External Notification this is disabled

Are you saying that the GNSS should work fine in `Slot D` or am I missing a setting somewhere?

ianmcorvidae, 2024-07-24:
You need to set the button GPIO to something; unset uses the default value of 9, which corresponds to IO5, which is within the slot D connector and will end up with GPS data getting sent to it and resetting your node. This is why the meshtastic docs at https://meshtastic.org/docs/hardware/devices/rak/peripherals/?rakmodules=GPS recommend using slot A. If you set that value to something else that isn't within the slot D connector (e.g., if you aren't using a button, it works to set it to 255 or something similarly inaccessible, as I understand it), then slot D works fine.

djekl, 2024-07-24:
Thank you! That's really helpful.

Will give this a go and see if that resolves my issue.

closed 2024-07-24 by thebentern

djekl, 2024-07-24:
Personally I wouldn't consider this issue closed, as I haven't found if thats fixing my issue. Also, why is the GPIO User Button mapped to Slot D, when the user expandable GPIO for the RAK board is on the IO slot.

Maybe the defaults for this board type need changed, and re-mapped by default to not cause a default conflict between manufacturer documentation, and Meshtastic documentation/defaults?

djekl, 2024-07-24:
Also, does anyone have any clear documentation as to which GPIO pins are mapped to which SLOTS, as to not cause an issue elsewhere.

garthvh, 2024-07-24:
If you are using any slot other than A for the gps you need to check the data sheet for each module and slot.

garthvh, 2024-07-24:
> Also, does anyone have any clear documentation as to which GPIO pins are mapped to which SLOTS, as to not cause an issue elsewhere.

RAK has excellent documentation, data sheets are available on their site.

markbirss, 2024-07-28:
> Also, does anyone have any clear documentation as to which GPIO pins are mapped to which SLOTS, as to not cause an issue elsewhere.

Check out the "RAK Pin Mapper" - https://docs.rakwireless.com/Knowledge-Hub/Pin-Mapper/