meshtastic/firmware #4329: [Bug]: Meshtastic 2.4.0 has requirements out of normal Ubuntu 22.04.2

Xaositek, 2024-07-25:
### Category

Other

### Hardware

DIY

### Firmware Version

2.4.0.46d7b82

### Description

Working on installing Meshtastic 2.4.0 onto my Raspberry Pi 3 with a Waveshare SX1262 hat. Went through the documentation (https://meshtastic.org/docs/hardware/devices/linux-native-hardware/) and performed all the steps.

_Tested Devices: Ubuntu 22.04 X86_64 with a CH341-SX1262 DIY dongle._

_user@pitastic:/etc/systemd/system$ lsb_release -a_
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.4 LTS
Release:	22.04
Codename:	jammy

_user@pitastic:/etc/meshtasticd$ sudo apt install openssl libssl-dev libulfius-dev liborcania-dev_
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
**liborcania-dev is already the newest version (2.2.1-1).**
libulfius-dev is already the newest version (2.7.7-1).
libssl-dev is already the newest version (3.0.2-0ubuntu1.16).
openssl is already the newest version (3.0.2-0ubuntu1.16).
0 upgraded, 0 newly installed, 0 to remove and 4 not upgraded.

### Relevant log output

```Shell
Jul 25 08:29:09 pitastic systemd[1]: Started Meshtastic Native Daemon.
Jul 25 08:29:09 pitastic meshtasticd[1107]: /usr/sbin/meshtasticd: error while loading shared libraries: liborcania.so.2.3: cannot open shared object file: No such file or directory
Jul 25 08:29:09 pitastic systemd[1]: meshtasticd.service: Main process exited, code=exited, status=127/n/a
Jul 25 08:29:09 pitastic systemd[1]: meshtasticd.service: Failed with result 'exit-code'.
```

madeofstown, 2024-07-26:
I believe if you build the source code the binary should run. The .deb package really only works well on RaspberryPi OS 64 bit.

jp-bennett, 2024-07-29:
We build for Debian Bookworm only at this point, as that's the source for the up-to-date Raspberry Pi image.

closed 2024-07-29 by jp-bennett

shmorgan, 2024-10-06:
Just  double checking, the wiki shows using Ubuntu 22.04, I used 22.04.5 LTS and ran into this issue too. This version is the bookworm version of the kernel but for some reason, Ubuntu does not have the 2.3 version of the library available for this version, so the Wiki may need to be updated to what flavor of OS we should use. Not sure which Ubuntu version has this library as part of the Library list. Wasted about an hour and a half going down this road.

markbirss, 2024-10-07:
> Just double checking, the wiki shows using Ubuntu 22.04, I used 22.04.5 LTS and ran into this issue too. This version is the bookworm version of the kernel but for some reason, Ubuntu does not have [...]

That is only for the USB dongle (DIY CH341 SX1262) as it is compiled using pio not installed from deb