meshtastic/firmware #4332: Rak4631 dead[Bug]:

DaveD58, 2024-07-26:
### Category

Hardware Compatibility

### Hardware

Rak4631

### Firmware Version

2.315.deb7c27

### Description

I lost (2) Rak4631 devices after loading the current available firmware. I reached out to Rak for help and they said it was a firmware induced failure?
1st unit lost all configuration settings including my Bluetooth password.  Was able to get back in with the default password and rest the configuration parameters. It worked for a day or so and now is completely dead. Tried to download new firmware via USB, appeared to accept the download but still nonfunctional. 2nd unit did basically the same thing, first the Bluetooth failed to connect, even with default password.  Tried to download new firmware via USB, seemed to take the file, but unit still nonfunctional.
Anyone else experience this?
Ideas or suggestions?
Thanks...

garthvh, 2024-07-26:
Do the factory reset process and re-flash the devices, really unlikely they are dead.

closed 2024-07-26 by garthvh

DaveD58, 2024-07-26:
What is the process for a factory reset?
Thanks
Dave

garthvh, 2024-07-26:
https://meshtastic.org/docs/getting-started/flashing-firmware/nrf52/nrf52-erase/

thebentern, 2024-07-26:
Easiest way to do this is use the web flasher and click the trash can icon after selecting the RAK-4631

https://flasher.meshtastic.org/

DaveD58, 2024-07-26:
Thanks,
I finally got the reset to work on one unit and it appears to be
functioning again.
The other one is still not working. It appears to do the reset and accept
the new firmware but I can't see it on Bluetooth. Red led is on steady and
the green one is blinking dimly, not bright like on the functional unit.
Thanks for your help and patience!
Dave

Archantium, 2024-09-15:
Same thing happened to a friend with two units. but it was with 2.5.0, send him this issue so he can try it. double tapping the button does nothing.

Archantium, 2024-09-15:
Currently trying to reproduce it on my device (have the same hardware)

Archantium, 2024-09-15:
So far mine still works, but he said it was working for a while after update and then both died, so I guess its now time to wait.

He only changed the device long and short name after the update

0181505050, 2024-09-15:
Yea that person is me, i tried grounding the UART2_TX pin to GND and see if i could get it seen by the PC, but still have the exact same problem.

caveman99, 2024-09-17:
try recovering the bootloader with nrfutil (not uf2 drop). that shold work no matter what. https://meshtastic.org/docs/getting-started/flashing-firmware/nrf52/update-nrf52-bootloader/#method-2-using-adafruit-nrfutil

Archantium, 2024-09-17:
I tried it like written in the link did sadly not work. Then we ground UART2_TX to ground and when u press reset it flashes green once but does not change the outcome (also not pressing reset does not change it), do you by any chance know what else could be it? it just disconnects after 5 seconds and nothing from there sadly.

> try recovering the bootloader with nrfutil (not uf2 drop). that shold work no matter what. https://meshtastic.org/docs/getting-started/flashing-firmware/nrf52/update-nrf52-bootloader/#method-2-using-adafruit-nrfutil

> adafruit-nrfutil --verbose dfu serial --package "wiscore_rak4631_board_bootloader-0.4.3_s140_6.1.1.zip" -p /dev/ttyACM0 -b 115200 --singlebank --touch 1200
Upgrading target on /dev/ttyACM0 with DFU package /home/username/Desktop/recover_rak4631/wiscore_rak4631_board_bootloader-0.4.3_s140_6.1.1.zip. Flow control is disabled, Single bank, Touch 1200
Touched serial port /dev/ttyACM0

Failed to upgrade target. Error is: Serial port could not be opened on /dev/ttyACM0. Reason: [Errno 2] could not open port /dev/ttyACM0: [Errno 2] No such file or directory: '/dev/ttyACM0'
Traceback (most recent call last):
  File "/home/username/Desktop/recover_rak4631/venv/lib/python3.12/site-packages/serial/serialposix.py", line 322, in open
    self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/dev/ttyACM0'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/username/Desktop/recover_rak4631/venv/lib/python3.12/site-packages/nordicsemi/dfu/dfu_transport_serial.py", line 113, in open
    self.serial_port = Serial(port=self.com_port, baudrate=self.baud_rate, rtscts=self.flow_control, timeout=self.timeout)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/username/Desktop/recover_rak4631/venv/lib/python3.12/site-packages/serial/serialutil.py", line 244, in __init__
    self.open()
  File "/home/username/Desktop/recover_rak4631/venv/lib/python3.12/site-packages/serial/serialposix.py", line 325, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 2] could not open port /dev/ttyACM0: [Errno 2] No such file or directory: '/dev/ttyACM0'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/username/Desktop/recover_rak4631/venv/lib/python3.12/site-packages/nordicsemi/__main__.py", line 296, in serial
    dfu.dfu_send_images()
  File "/home/username/Desktop/recover_rak4631/venv/lib/python3.12/site-packages/nordicsemi/dfu/dfu.py", line 226, in dfu_send_images
    self._dfu_send_image(HexType.SD_BL, self.manifest.softdevice_bootloader)
  File "/home/username/Desktop/recover_rak4631/venv/lib/python3.12/site-packages/nordicsemi/dfu/dfu.py", line 157, in _dfu_send_image
    self.dfu_transport.open()
  File "/home/username/Desktop/recover_rak4631/venv/lib/python3.12/site-packages/nordicsemi/dfu/dfu_transport_serial.py", line 115, in open
    raise NordicSemiException("Serial port could not be opened on {0}. Reason: {1}".format(self.com_port, e))
nordicsemi.exceptions.NordicSemiException: Serial port could not be opened on /dev/ttyACM0. Reason: [Errno 2] could not open port /dev/ttyACM0: [Errno 2] No such file or directory: '/dev/ttyACM0'

Possible causes:
- Selected Bootloader version does not match the one on Bluefruit device.
    Please upgrade the Bootloader or select correct version in Tools->Bootloader.
- Baud rate must be 115200, Flow control must be off.
- Target is not in DFU mode. Ground DFU pin and RESET and release both to enter DFU mode.

DaveD58, 2024-10-05:
My unit again locked up and apparently lost it's configuration.
After much to do and frustration, I believe I have finally successfully
loaded the latest version of software.
Wondering how long it will last this time....